openwrt/routing #622: luci-app-cjdns:  1.3-6 module 'luci.cbi' not found

shithead, 2020-10-27:
OpenWRT version: LuCI openwrt-19.07 branch (git-20.287.57033-3d52019) / OpenWrt 19.07.4 r11208-ce6496d796 

URL: ip.ad.dr.ess/cgi-bin/luci/admin/services/cjdns

backtrack:
> /usr/lib/lua/luci/dispatcher.lua:1334: module 'luci.cbi' not found:
> 	no field package.preload['luci.cbi']
> 	no file './luci/cbi.lua'
> 	no file '/usr/share/lua/luci/cbi.lua'
> 	no file '/usr/share/lua/luci/cbi/init.lua'
> 	no file '/usr/lib/lua/luci/cbi.lua'
> 	no file '/usr/lib/lua/luci/cbi/init.lua'
> 	no file './luci/cbi.so'
> 	no file '/usr/lib/lua/luci/cbi.so'
> 	no file '/usr/lib/lua/loadall.so'
> 	no file './luci.so'
> 	no file '/usr/lib/lua/luci.so'
> 	no file '/usr/lib/lua/loadall.so'
> stack traceback:
> 	[C]: in function 'require'
> 	/usr/lib/lua/luci/dispatcher.lua:1334: in function '_cbi'
> 	/usr/lib/lua/luci/dispatcher.lua:1023: in function 'dispatch'
> 	/usr/lib/lua/luci/dispatcher.lua:478: in function </usr/lib/lua/luci/dispatcher.lua:477>

`

wfleurant, 2021-01-02:
understood looking at this in #581

wfleurant, 2021-06-03:
I could not reproduce this bug in OpenWrt 21.02

closed 2021-06-03 by wfleurant