openwrt/routing #651: olsrd dependency on firewall3

champtar, 2021-04-02:
Hi @zioproto 

olsrd has an implicit dependency on firewall3 for the smartgw feature.
firewall4 is based on nftables and thus part of olsrd will not work with it.
If possible use procd firewall objects (exemples https://github.com/search?p=2&q=org%3Aopenwrt+json_add_array+firewall&type=Code) else consider adding:
- an explicit dependency on firewall3 until this is fixed
- some checks if ip/ip6tables exists or not

zioproto, 2021-04-05:
@fhuberts if I remember correctly you are familiar with the smartgw feature, correct ? Could you help here ? thanks

fhuberts, 2021-04-05:
It's been many years since I worked on olsrd.
But IIRC then the firewall dependend stuff is all in scripts, and those can be converted to nftables.
Might not be a trivial job but it can be done.
Not by me though ;-)

PolynomialDivision, 2021-06-09:
>  If possible use procd firewall objects (exemples https://github.com/search?p=2&q=org%3Aopenwrt+json_add_array+firewall&type=Code) else consider adding:

Could you do this? I think you just have to change that:
https://github.com/openwrt-routing/packages/blob/b769f1d7985a812e0b13c7d2598dc89f84e6a26e/olsrd/files/olsrd.sh#L745-L812