openwrt/routing #663: Move to openwrt/routing.git

aparcar, 2021-06-02:
Hi, I'd like to finally move this repository to openwrt/routing.git rather than keeping it under a different organization name.

For starters, having it under a different name messes up access rights, e.g. when developers want to branch for a new release.

Please grant me full access on this repository and I'll do the move. GitHub automatically handles forwarding of Git setups which still use the "old" address.

mwarning, 2021-06-03:
I remember there was a discussion about this. Can we link it here?

aparcar, 2021-06-03:
It was initially [here](https://github.com/openwrt-routing/packages/issues/184) which suggested to merge it into packages.git, something I'd still prefer but clearly accept the maintainers concerns (see thread). Since the move doesn't "fix" the linked issue I decided to open a new one.

mwarning, 2021-06-04:
I would be willing to give this a go. But I like to get a yes or no from the members of openwrt-routing:
@wfleurant, @oryon, @altergui, @simonwunderlich,  @sbyx, @jow-, @elektra42, @axn, @fingon, @ffainelli, @p4u, @acinonyx, @zioproto, @zorun, @kerneis, @mwarning, @lynxis, @lperkov, @bittorf, @lindnermarek, @nolith

If we get a majority, I will let @aparcar take care of the rest.

PolynomialDivision, 2021-06-04:
@mwarning you did both? xD

mwarning, 2021-06-04:
@PolynomialDivision it was meant as a visual hint (removed now) :-)

lindnermarek, 2021-06-04:
Moving the repository to openwrt/routing.git while preserving the current structure & permissions works for me.

Gave my thumbs up.

elektra42, 2021-06-04:
Fine for me, too.Regards,Elektra

PolynomialDivision, 2021-06-09:
> It was initially [here](https://github.com/openwrt-routing/packages/issues/184) which suggested to merge it into packages.git, something I'd still prefer but clearly accept the maintainers concerns (see thread). Since the move doesn't "fix" the linked issue I decided to open a new one.

I would also prefer merging it into packages feed. Just for the reason that this community has a lot of active reviewers. If you do a PR there, you instantly get feedback. It is always nice to have some other people watching on what you do even if you are maintainer.

lindnermarek, 2021-06-09:
> I would also prefer merging it into packages feed. Just for the reason that this community has a lot of active reviewers. If you do a PR there, you instantly get feedback. It is always nice to have some other people watching on what you do even if you are maintainer.

Please read the original conversation on this topic: https://github.com/openwrt-routing/packages/issues/184

aparcar, 2021-06-11:
It's been a week, would you kindly make me a repository owner?

aparcar, 2021-06-11:
I will no proceed moving this repository, first it will be renamed to `routing` and then moved to `openwrt/routing`.

wfleurant, 2021-06-11:
boh!

Sent from ProtonMail mobile

-------- Original Message --------

aparcar, 2021-06-11:
I finished the move, all current maintainers are invited as admins to the new repository.

closed 2021-06-11 by aparcar

wfleurant, 2021-06-11:
wicked glad to see this is happen'in! cheers all,

Sent from ProtonMail mobile

-------- Original Message --------

mwarning, 2021-06-11:
Thanks @aparcar for doing this.

simonwunderlich, 2021-08-20:
@aparcar I don't have access to the new repo, can you please check? I would like to merge the batman-adv merge requests. thank you!

aparcar, 2021-08-20:
I've added you.