openwrt/routing #665: Batman compile failed for aarch64

Mon-ius, 2021-06-03:
```
2021-06-03T07:09:41.1002834Z Applying ./patches/0001-Revert-batman-adv-genetlink-move-to-smaller-ops-wher.patch using plaintext: 
2021-06-03T07:09:41.1019804Z patching file compat-include/net/genetlink.h
2021-06-03T07:09:41.1022975Z patching file net/batman-adv/netlink.c
2021-06-03T07:09:41.1028712Z 
2021-06-03T07:09:41.1031088Z Applying ./patches/0002-Revert-batman-adv-Add-new-include-for-min-max-helper.patch using plaintext: 
2021-06-03T07:09:41.1049821Z patching file compat-include/linux/minmax.h
2021-06-03T07:09:41.8325454Z patching file net/batman-adv/bat_v.c
2021-06-03T07:09:41.8326792Z patching file net/batman-adv/bat_v_elp.c
2021-06-03T07:09:41.8327850Z patching file net/batman-adv/bat_v_ogm.c
2021-06-03T07:09:41.8329231Z patching file net/batman-adv/fragmentation.c
2021-06-03T07:09:41.8330341Z patching file net/batman-adv/hard-interface.c
2021-06-03T07:09:41.8331649Z patching file net/batman-adv/main.c
2021-06-03T07:09:41.8332645Z patching file net/batman-adv/netlink.c
2021-06-03T07:09:41.8334050Z patching file net/batman-adv/tp_meter.c
2021-06-03T07:09:41.8334653Z 
2021-06-03T07:09:41.8336685Z Applying ./patches/0003-batman-adv-Fix-build-of-multicast-code-against-Linux.patch using plaintext: 
2021-06-03T07:09:41.8338308Z patching file net/batman-adv/multicast.c
2021-06-03T07:09:41.8338664Z 
2021-06-03T07:09:41.8340006Z Applying ./patches/0004-batman-adv-Always-send-iface-index-name-in-genlmsg.patch using plaintext: 
2021-06-03T07:09:41.8341450Z patching file net/batman-adv/bat_iv_ogm.c
2021-06-03T07:09:41.8342127Z patching file net/batman-adv/bat_v.c
2021-06-03T07:09:41.8342807Z patching file net/batman-adv/netlink.c
2021-06-03T07:09:41.8344436Z touch /home/runner/work/auto-bbr-v2/auto-bbr-v2/lede/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/batman-adv-2021.1/.prepared_6ad2801ba9b1ccd6a112c75d211ce86f_18f1e190c5d53547fed41a3eaa76e9e9
2021-06-03T07:09:41.8346427Z rm -f /home/runner/work/auto-bbr-v2/auto-bbr-v2/lede/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/batman-adv-2021.1/.configured_*
2021-06-03T07:09:41.8347956Z rm -f /home/runner/work/auto-bbr-v2/auto-bbr-v2/lede/staging_dir/target-aarch64_cortex-a72_musl/stamp/.batman-adv_installed
2021-06-03T07:09:41.8369297Z (cd /home/runner/work/auto-bbr-v2/auto-bbr-v2/lede/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/batman-adv-2021.1/./; if [ -x ./configure ]; then find /home/runner/work/auto-bbr-v2/auto-bbr-v2/lede/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/batman-adv-2021.1/ -name config.guess | xargs -r chmod u+w; find /home/runner/work/auto-bbr-v2/auto-bbr-v2/lede/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/batman-adv-2021.1/ -name config.guess | xargs -r -n1 cp --remove-destination /home/runner/work/auto-bbr-v2/auto-bbr-v2/lede/scripts/config.guess; find /home/runner/work/auto-bbr-v2/auto-bbr-v2/lede/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/batman-adv-2021.1/ -name config.sub | xargs -r chmod u+w; find /home/runner/work/auto-bbr-v2/auto-bbr-v2/lede/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/batman-adv-2021.1/ -name config.sub | xargs -r -n1 cp --remove-destination /home/runner/work/auto-bbr-v2/auto-bbr-v2/lede/scripts/config.sub; AR="aarch64-openwrt-linux-musl-gcc-ar" AS="aarch64-openwrt-linux-musl-gcc -c -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -ffile-prefix-map=/home/runner/work/auto-bbr-v2/auto-bbr-v2/lede/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/batman-adv-2021.1=batman-adv-2021.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD=aarch64-openwrt-linux-musl-ld NM="aarch64-openwrt-linux-musl-gcc-nm" CC="aarch64-openwrt-linux-musl-gcc" GCC="aarch64-openwrt-linux-musl-gcc" CXX="aarch64-openwrt-linux-musl-g++" RANLIB="aarch64-openwrt-linux-musl-gcc-ranlib" STRIP=aarch64-openwrt-linux-musl-strip OBJCOPY=aarch64-openwrt-linux-musl-objcopy OBJDUMP=aarch64-openwrt-linux-musl-objdump SIZE=aarch64-openwrt-linux-musl-size CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -ffile-prefix-map=/home/runner/work/auto-bbr-v2/auto-bbr-v2/lede/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/batman-adv-2021.1=batman-adv-2021.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " CXXFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -ffile-prefix-map=/home/runner/work/auto-bbr-v2/auto-bbr-v2/lede/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/batman-adv-2021.1=batman-adv-2021.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " CPPFLAGS="-I/home/runner/work/auto-bbr-v2/auto-bbr-v2/lede/staging_dir/target-aarch64_cortex-a72_musl/usr/include -I/home/runner/work/auto-bbr-v2/auto-bbr-v2/lede/staging_dir/toolchain-aarch64_cortex-a72_gcc-8.4.0_musl/usr/include -I/home/runner/work/auto-bbr-v2/auto-bbr-v2/lede/staging_dir/toolchain-aarch64_cortex-a72_gcc-8.4.0_musl/include/fortify -I/home/runner/work/auto-bbr-v2/auto-bbr-v2/lede/staging_dir/toolchain-aarch64_cortex-a72_gcc-8.4.0_musl/include " LDFLAGS="-L/home/runner/work/auto-bbr-v2/auto-bbr-v2/lede/staging_dir/target-aarch64_cortex-a72_musl/usr/lib -L/home/runner/work/auto-bbr-v2/auto-bbr-v2/lede/staging_dir/target-aarch64_cortex-a72_musl/lib -L/home/runner/work/auto-bbr-v2/auto-bbr-v2/lede/staging_dir/toolchain-aarch64_cortex-a72_gcc-8.4.0_musl/usr/lib -L/home/runner/work/auto-bbr-v2/auto-bbr-v2/lede/staging_dir/toolchain-aarch64_cortex-a72_gcc-8.4.0_musl/lib -znow -zrelro "   ./configure --target=aarch64-openwrt-linux --host=aarch64-openwrt-linux --build=x86_64-pc-linux-gnu --program-prefix="" --program-suffix="" --prefix=/usr --exec-prefix=/usr --bindir=/usr/bin --sbindir=/usr/sbin --libexecdir=/usr/lib --sysconfdir=/etc --datadir=/usr/share --localstatedir=/var --mandir=/usr/man --infodir=/usr/info --disable-nls  ; fi; )
2021-06-03T07:09:41.8388383Z touch /home/runner/work/auto-bbr-v2/auto-bbr-v2/lede/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/batman-adv-2021.1/.configured_68b329da9893e34099c7d8ad5cb9c940
2021-06-03T07:09:41.8390254Z rm -f /home/runner/work/auto-bbr-v2/auto-bbr-v2/lede/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/batman-adv-2021.1/.built
2021-06-03T07:09:41.8391821Z touch /home/runner/work/auto-bbr-v2/auto-bbr-v2/lede/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/batman-adv-2021.1/.built_check
2021-06-03T07:09:41.8394977Z cat /dev/null > /home/runner/work/auto-bbr-v2/auto-bbr-v2/lede/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/symvers/batman-adv.symvers; for subdir in net/batman-adv; do cat /home/runner/work/auto-bbr-v2/auto-bbr-v2/lede/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/symvers/*.symvers 2>/dev/null > /home/runner/work/auto-bbr-v2/auto-bbr-v2/lede/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/batman-adv-2021.1/$subdir/Module.symvers; done
2021-06-03T07:09:41.8410105Z make -j1 -C "/home/runner/work/auto-bbr-v2/auto-bbr-v2/lede/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/linux-5.4.123" KCFLAGS="-ffile-prefix-map=/home/runner/work/auto-bbr-v2/auto-bbr-v2/lede/build_dir/target-aarch64_cortex-a72_musl=target-aarch64_cortex-a72_musl" HOSTCFLAGS="-O2 -I/home/runner/work/auto-bbr-v2/auto-bbr-v2/lede/staging_dir/host/include -I/home/runner/work/auto-bbr-v2/auto-bbr-v2/lede/staging_dir/hostpkg/include -I/home/runner/work/auto-bbr-v2/auto-bbr-v2/lede/staging_dir/target-aarch64_cortex-a72_musl/host/include -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="aarch64-openwrt-linux-musl-" ARCH="arm64" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="builder" KBUILD_BUILD_HOST="buildhost" KBUILD_BUILD_TIMESTAMP="Thu Jun  3 02:04:37 2021" KBUILD_BUILD_VERSION="0" HOST_LOADLIBES="-L/home/runner/work/auto-bbr-v2/auto-bbr-v2/lede/staging_dir/host/lib" KBUILD_HOSTLDLIBS="-L/home/runner/work/auto-bbr-v2/auto-bbr-v2/lede/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls= KERNELRELEASE=5.4.123 M="/home/runner/work/auto-bbr-v2/auto-bbr-v2/lede/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/batman-adv-2021.1/net/batman-adv" CONFIG_BATMAN_ADV=m CONFIG_BATMAN_ADV_DEB [... cut ...]
2021-06-03T07:09:47.7620451Z [command]/usr/bin/git config --local --name-only --get-regexp http\.https\:\/\/github\.com\/\.extraheader
2021-06-03T07:09:47.7661953Z http.https://github.com/.extraheader
2021-06-03T07:09:47.7673276Z [command]/usr/bin/git config --local --unset-all http.https://github.com/.extraheader
2021-06-03T07:09:47.7718628Z [command]/usr/bin/git submodule foreach --recursive git config --local --name-only --get-regexp 'http\.https\:\/\/github\.com\/\.extraheader' && git config --local --unset-all 'http.https://github.com/.extraheader' || :
2021-06-03T07:09:47.8040166Z Cleaning up orphan processes
```

ecsv, 2021-06-04:
Please fix the topic

---

Something is broken with your mac80211 package. I have to assume this because it will provide netif_rx_any_context when you are using kernel 5.4 - basically for everything not >=5.10

I personally cannot reproduce this problem on master - also the buildbots cannot reproduce it for OpenWrt 21.02 (as seen in https://downloads.openwrt.org/releases/21.02.0-rc2/targets/ath79/generic/kmods/5.4.119-1-4f1271a56f0192cef6cc5531235d2d0e/ )

Here is my (diff)config for master:

    CONFIG_TARGET_bcm27xx=y
    CONFIG_TARGET_bcm27xx_bcm2711=y
    CONFIG_TARGET_bcm27xx_bcm2711_DEVICE_rpi-4=y
    CONFIG_BATMAN_ADV_BATMAN_V=y
    CONFIG_BATMAN_ADV_BLA=y
    CONFIG_BATMAN_ADV_DAT=y
    CONFIG_BATMAN_ADV_MCAST=y
    CONFIG_PACKAGE_batctl-default=y
    CONFIG_PACKAGE_kmod-batman-adv=y
    CONFIG_PACKAGE_kmod-crypto-crc32c=y
    CONFIG_PACKAGE_kmod-crypto-hash=y
    CONFIG_PACKAGE_kmod-lib-crc16=y
    CONFIG_PACKAGE_kmod-lib-crc32c=y

ecsv, 2021-06-04:
I saw that you posted other issues (https://github.com/coolsnowwolf/lede/issues/6921) about OpenWrt build errors and used https://github.com/coolsnowwolf/lede

If you really use a non-official OpenWrt repository full a weird hacks+outdated packages and then start to report build problems at openwrt (feed) repositories - please don't do this. OpenWrt can be found at

* https://git.openwrt.org/?p=openwrt/openwrt.git;a=summary
* https://github.com/openwrt/openwrt

Whatever this coolsnowwolf/lede stuff is, it is not a supported OpenWrt version and I will not support random hacks.

Mon-ius, 2021-06-04:
> I saw that you posted other issues ([coolsnowwolf/lede#6921](https://github.com/coolsnowwolf/lede/issues/6921)) about OpenWrt build errors and used https://github.com/coolsnowwolf/lede
> 
> If you really use a non-official OpenWrt repository full a weird hacks+outdated packages and then start to report build problems at openwrt (feed) repositories - please don't do this. OpenWrt can be found at
> 
> * https://git.openwrt.org/?p=openwrt/openwrt.git;a=summary
> * https://github.com/openwrt/openwrt
> 
> Whatever this coolsnowwolf/lede stuff is, it is not a supported OpenWrt version and I will not support random hacks.

Yes, I fixed this one for RPi4, see https://github.com/coolsnowwolf/packages/pull/154, But didn't pass the bot for LEDE. I am familiar with CMAKE and Make, but kind of new to the OpenWrt community. Technically, I wondering why using the same feeds can cause thus error that related to Linux version even it was specified in the `.config` file. Also, I tried to compile the `routing` feeds only but still leads to the error.

Mon-ius, 2021-06-04:
> Please fix the topic
>
> Something is broken with your mac80211 package. I have to assume this because it will provide netif_rx_any_context when you are using kernel 5.4 - basically for everything not [...]

I can remember I used such config once before I turned to use LEDE. Now I will run it again with Openwrt, see https://github.com/Mon-ius/auto-bbr-v2/actions/runs/906415403.

The core things are:
```
curl -L https://downloads.openwrt.org/releases/21.02.0-rc2/targets/bcm27xx/bcm2711/config.buildinfo -o .config
./scripts/feeds update -a && ./scripts/feeds install -a
make defconfig
make download -j8
make -j8 V=s
```

ecsv, 2021-06-04:
> Yes, I fixed this one for RPi4, see [coolsnowwolf/packages#154](https://github.com/coolsnowwolf/packages/pull/154), But didn't pass the bot for LEDE. I am familiar with CMAKE and Make, but kind of new to the OpenWrt community.

Whatever this coolsnowwolf is doing is neither LEDE nor OpenWrt. LEDE was a fork of OpenWrt at some point around 2017 but is now merged back to OpenWrt - they are now the same project again and use the name OpenWrt. Whatever coolsnowwolf is having there is neither and you have to discuss whatever problems you have with it with coolsnowwolf.

> Technically, I wondering why using the same feeds can cause thus error that related to Linux version even it was specified in the `.config` file. Also, I tried to compile the `routing` feeds only but still leads to the error.

The official feeds for OpenWrt are made for specific versions of OpenWrt. For example, batman-adv from openwrt-routing's openwrt-21.02 branch is made with the OpenWrt 21.02 packages in mind (compiler, linux, mac80211, ...). These packages are at a specific version and provide various compile and runtime functionality. So it is rather likely that it will not work as expected when you are trying to use this OpenWrt specific feed against something which is not providing the correct version of the OpenWrt packages.

> Now I will run it again with Openwrt, see https://github.com/Mon-ius/auto-bbr-v2/actions/runs/906415403.

No idea what you want to show me here but batman-adv build perfectly file. The only error I see is caused by your workflow because you add something with an invalid ("rpi-4") to the GITHUB_ENV. See for example https://github.community/t/github-env-invalid-environment-variable-format/168645

Mon-ius, 2021-06-04:
Yep, some errors for not setting perfectly. But don't mind that. Let's focus on batman-dev, if using the same feeds, what outside environment could influence the <linux/version.h> file. And if the main source code consider it very well, as you described 'batman-adv build perfectly file'. Technically, if it is possible that dependency in 
`DEPENDS:=+BATMAN_ADV_BLA:kmod-lib-crc16 +kmod-lib-crc32c +kmod-cfg80211 +batctl` caused such things.

ecsv, 2021-06-05:
* batman-adv package from openwrt-21.02 branch builds perfectly fine against OpenWrt 21.02
* batman-adv package from openwrt-19.07 branch builds perfectly fine against OpenWrt 19.07
* batman-adv package from master branch builds perfectly fine against OpenWrt master branch

So it is working as expected.

As explained before: The problem that the batman-adv package from the master branch is not building against coolsnowwolf whatever repo is caused by coolsnowwolf whatever repo's mac80211 package. Tthe mac80211 compat module (part of kmod-cfg80211) in coolsnowwolf's whatever repo is not up to date and not matching the version in OpenWrt maser. Please discuss this with coolsnowwolf.

closed 2021-06-05 by simonwunderlich