openwrt/routing #677: kmod-batman-adv: vlan ap isolation setting is ignored upon network restart/reload

dccharacter, 2021-06-09:
Maintainer: @simonwunderlich
Environment:
```
[    0.000000] Linux version 4.14.221 (builder@buildhost) (gcc version 7.5.0 (Op
enWrt GCC 7.5.0 r11306-c4a6851c72)) #0 SMP Mon Feb 15 15:22:37 2021
[    0.000000] CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c5387d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instructio
n cache
[    0.000000] OF: fdt: Machine model: ASUS RT-AC58U
```

`OpenWrt 19.07.7, r11306-c4a6851c72`

```
root@asus:~# opkg list_installed |grep batman
kmod-batman-adv - 4.14.221+2019.2-11
```

Description:

ap_isolation for vlans is not preserved through network reset/reload and device reboot. `ap_isolation '1'` for `bat0.xxxx` in `/etc/config/network' does not have effect.
```
root@asus:~# batctl -m bat0.4 ap
disabled
root@asus:~# batctl -m bat0.4 ap 1
root@asus:~# batctl -m bat0.4 ap
enabled
root@asus:~# service network restart
root@asus:~# batctl -m bat0.4 ap
disabled
root@asus:~# cat /etc/config/network |grep -A 11 bat0.4
        option ifname 'bat0.4'
        option delegate '0'
        option proto 'none'
#       option proto 'static'
#       option ipaddr '192.168.40.11'
#       option gateway '192.168.40.1'
#       list dns '192.168.40.1'
#       option netmask '255.255.255.0'
        option type 'bridge'
        option stp '1'
        option ap_isolation '1'
```

ecsv, 2021-06-27:
Looks to me like you are doing it wrong. Please have a look at the examples from https://www.open-mesh.org/projects/batman-adv/wiki/Batman-adv-openwrt-config

It clearly states that the ap_isolation option is only relevant for the proto batadv_vlan. But you are using here proto none.

@simonwunderlich Please close this ticket as invalid.

dccharacter, 2021-07-18:
I tried changing the proto to batadv_vlan, but no effect. If I'm reading
the source code correctly, the ap_siolation is hardcoded and is never read
from config or modified later in the process.

ecsv, 2021-07-18:
> I tried changing the proto to batadv_vlan, but no effect. If I'm reading

Just tried it myself with following config in /etc/config/network:

    config interface 'bat0'
        option proto 'batadv'
        option ap_isolation '1'
    
    config interface 'bat0_hardif_eth0'
        option proto 'batadv_hardif'
        option master 'bat0'
        option mtu '1536'
        option ifname 'eth0'
    
    config interface 'bat0_vlan1'
        option proto 'batadv_vlan'
        option ifname 'bat0.1'
        option ap_isolation '1'
    
    config interface 'bat0_vlan2'                                                            
        option proto 'batadv_vlan'          
        option ifname 'bat0.2'
        option ap_isolation '0'

This is what I get with this config

    root@OpenWrt:/# /etc/init.d/network restart
    root@OpenWrt:/# batctl meshif bat0 ap_isolation
    enabled
    root@OpenWrt:/# batctl vlan bat0.1 ap_isolation
    enabled
    root@OpenWrt:/# batctl vlan bat0.2 ap_isolation
    disabled

> the source code correctly, the ap_siolation is hardcoded and is never read
> from config or modified later in the process.
 
Incorrect, see: https://github.com/openwrt/routing/blob/8483b6d99fc6fb2d9c5917892d4493b1b2caaf27/batman-adv/files/lib/netifd/proto/batadv_vlan.sh#L18

And for non-vlan devices:

* read from config via: https://github.com/openwrt/routing/blob/8483b6d99fc6fb2d9c5917892d4493b1b2caaf27/batman-adv/files/lib/netifd/proto/batadv.sh#L55
* set via https://github.com/openwrt/routing/blob/8483b6d99fc6fb2d9c5917892d4493b1b2caaf27/batman-adv/files/lib/netifd/proto/batadv.sh#L78

@simonwunderlich  please close this issue as invalid.

closed 2021-11-19 by simonwunderlich