openwrt/routing #679: bird2: Crashes on IPQ806X when using OSPF

mattreeve, 2021-06-11:
Maintainer: @\tohojo
Environment: (Qualcomm Atheros IPQ806X, NETGEAR Nighthawk X4S R7800, 21.02-rc2)

Description:

If ospf v2 or v3 is configured in bird.conf, it fails to start with this error:

Fri Jun 11 14:41:11 2021 daemon.info bird: Started
Fri Jun 11 14:41:11 2021 kern.err kernel: [ 3500.853248] Alignment trap: not handling instruction f44c0a1f at [<00035848>] Fri Jun 11 14:41:11 2021 kern.alert kernel: [ 3500.853283] 8<--- cut here ---
Fri Jun 11 14:41:11 2021 kern.alert kernel: [ 3500.859363] Unhandled fault: alignment exception (0x801) at 0x007e0624
Fri Jun 11 14:41:11 2021 kern.alert kernel: [ 3500.862443] pgd = 0bbef4fd
Fri Jun 11 14:41:11 2021 kern.alert kernel: [ 3500.868821] [007e0624] *pgd=5d6ca835, *pte=5c40b75f, *ppte=5c40bc7f

If ospf is removed from bird.conf, then bird starts fine.

It seems to be something to do with a struct not being properly aligned on the ARMv7 architecture. Note that both bird1-ipv4 and bird1-ipv6 run OK.

The problem also exists on this hardware with other versions, such as bird 2.0.7 on openwrt 19.07.7.

mattreeve, 2021-07-30:
I've identified the problem and reported it upstream, with a potential fix https://bird.network.cz/pipermail/bird-users/2021-June/015545.html

tohojo, 2021-08-09:
Care to turn that into a patch and open a PR - we can use that until it gets fixed upstream...

mattreeve, 2021-08-11:
PR #712 created for 21.02. Do you want one for master, or should we wait a while to see if upstream fix it first?

mattreeve, 2021-08-13:
Fixed in master and 21.02, pending a proper fix coming from upstream.

closed 2021-08-13 by mattreeve