openwrt/routing #682: Delete Stale branches ?

PolynomialDivision, 2021-06-11:
Can we delete all the branches except 19.07, 21.02 and master?

mwarning, 2021-06-11:
I would also keep "openwrt-18.06". But let's wait for input from others, we might not know if any branches are signification in some way.

mwarning, 2021-06-11:
I think we can delete the branches marek. (ping @lindnermarek)
Not sure about the purpose of the for-* branches.

PolynomialDivision, 2021-06-11:
Okay. Let's put a deadline? If there is no nack until next Friday we delete them? I like to tidy up this repro. :) The Readme looks much nicer now. Maybe we can even go for a cool logo or stuff like this?

mwarning, 2021-06-11:
Ok. Also let's also keep lede-17.01, as it might be in use.
@lynxis do you remember what the purpose ofthe for-* branches is?

lindnermarek, 2021-06-12:
> I think we can delete the branches marek. (ping @lindnermarek)
> Not sure about the purpose of the for-* branches.

The for-* branches were branches to support older Openwrt releases. Those are clearly obsolete by now and can be removed if you ask me.

SvenRoederer, 2021-06-12:
> @lynxis do you remember what the purpose of the for-* branches is?

Maybe you remember a time before LEDE split and rejoin when there was "Attitude Adjustment", "Barrier Breaker", "Chaos Calmer" releases? These branches relate to the these old releases, like https://github.com/openwrt/packages/tree/for-14.07 is doing.
I suggest to keep them for historic builds or to tag them as "for-xx.yy_EOL" (I don't think anyone will contribute to them anymore)

jow-, 2021-06-12:
As written above, the `for-*` branches were for older OpenWrt releases, before the `lede-yy.mm` and later `openwrt-yy.mm` naming schema was adopted. I'd like to retain them, even if it is just for historic reference.

zorun, 2021-06-13:
+1 to keep old release branches for historical reference, I don't see what we would gain by dropping them.

PolynomialDivision, 2021-06-13:
Okay, I think we can close this issue and leave everything like it is? ;)

mwarning, 2021-06-13:
Well, there are still the branches from marek. @lindnermarek do you need them?

PolynomialDivision, 2021-08-29:
I think we can close?

mwarning, 2021-08-29:
Yes, I think we can close this ticket.
@lindnermarek please have a look at your branches.

closed 2021-08-29 by mwarning