openwrt/routing #691: olsrd: not starting with procd - as no interface configured

SvenRoederer, 2021-06-17:
Maintainer: @PolynomialDivision 
Environment: OpenWrt-21.02.0-rc3; TPlink WDR3600

Description: 

I recently build a image with all changes from OpenWrt-21.02.0-rc2 to OpenWrt-21.02.0-rc3 (which includes the recent changes of olsrd). With only these changes olsrd fails to start initially, but after manually restarting it later it's working.
In the attached system-log there is a message about `olsrd exit: main: Bad configuration`. 
In the attached logfiles you can see, that during 1st-start (bootup) there is no interface defined. In the config after the 2nd-start (manually triggered) both expected interfaces are defined.
The parallel configuration of the olsrd6 is working fine in the 1st start.

Looks like a race condition with bringing up the interfaces.

[system-log.txt](https://github.com/openwrt/routing/files/6672337/system-log.txt)
[olsr_1ststart.conf.txt](https://github.com/openwrt/routing/files/6672347/olsr_1ststart.conf.txt)
[olsr_2ndstart.conf.txt](https://github.com/openwrt/routing/files/6672348/olsr_2ndstart.conf.txt)

PolynomialDivision, 2021-06-18:
Thanks for reporting! I pushed some change that should fix this. I also need to test it.

SvenRoederer, 2021-06-19:
the change aff5fc9 did not improve really, looks essentially the same:

~~~
Tue Apr 20 02:42:00 2021 daemon.notice hostapd: wlan1-dhcp-5: interface state COUNTRY_UPDATE->ENABLED
Tue Apr 20 02:42:00 2021 daemon.notice hostapd: wlan1-dhcp-5: AP-ENABLED
Tue Apr 20 02:42:00 2021 daemon.notice hostapd: wlan0-dhcp-2: interface state COUNTRY_UPDATE->ENABLED
Tue Apr 20 02:42:00 2021 daemon.notice hostapd: wlan0-dhcp-2: AP-ENABLED
Tue Apr 20 02:42:03 2021 kern.info kernel: [   43.779782] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0-mesh-2: link becomes ready
Tue Apr 20 02:42:03 2021 kern.info kernel: [   43.844885] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1-mesh-5: link becomes ready
Tue Apr 20 02:42:03 2021 daemon.info olsrd: /etc/rc.d/S65olsrd: olsrd_write_interface() Warning: Interface 'wireless0' not found, skipped
Tue Apr 20 02:42:03 2021 daemon.notice procd: /etc/rc.d/S65olsrd: olsrd: /etc/rc.d/S65olsrd: olsrd_write_interface() Warning: Interface 'wireless0' not found, skip
ped
Tue Apr 20 02:42:03 2021 daemon.info olsrd: /etc/rc.d/S65olsrd: olsrd_write_interface() Warning: Interface 'wireless1' not found, skipped
Tue Apr 20 02:42:03 2021 daemon.notice procd: /etc/rc.d/S65olsrd: olsrd: /etc/rc.d/S65olsrd: olsrd_write_interface() Warning: Interface 'wireless1' not found, skip
ped
Tue Apr 20 02:42:03 2021 daemon.notice netifd: Network device 'wlan0-mesh-2' link is up
Tue Apr 20 02:42:03 2021 daemon.notice netifd: Interface 'wireless0' is enabled
Tue Apr 20 02:42:03 2021 daemon.notice netifd: Interface 'wireless0' is setting up now
Tue Apr 20 02:42:04 2021 daemon.notice netifd: Interface 'wireless0' is now up
Tue Apr 20 02:42:04 2021 daemon.notice netifd: Interface 'wireless0' has link connectivity
Tue Apr 20 02:42:04 2021 daemon.notice netifd: Network device 'wlan0-dhcp-2' link is up
Tue Apr 20 02:42:04 2021 daemon.notice netifd: Network device 'wlan1-mesh-5' link is up
Tue Apr 20 02:42:04 2021 daemon.notice netifd: Interface 'wireless1' is enabled
Tue Apr 20 02:42:04 2021 daemon.notice netifd: Interface 'wireless1' is setting up now
Tue Apr 20 02:42:04 2021 daemon.notice netifd: Interface 'wireless1' is now up
Tue Apr 20 02:42:04 2021 daemon.notice netifd: Interface 'wireless1' has link connectivity
Tue Apr 20 02:42:04 2021 daemon.notice netifd: Network device 'wlan1-dhcp-5' link is up
Tue Apr 20 02:42:04 2021 daemon.info olsrd: /etc/rc.d/S65olsrd: olsrd_setup_smartgw_rules() Notice: Inserting firewall rules for SmartGateway
Tue Apr 20 02:42:04 2021 daemon.notice procd: /etc/rc.d/S65olsrd: olsrd: /etc/rc.d/S65olsrd: olsrd_setup_smartgw_rules() Notice: Inserting firewall rules for Smart
Gateway
Tue Apr 20 02:42:04 2021 user.notice firewall: Reloading firewall due to ifup of tunl0 (tunl0)
Tue Apr 20 02:42:05 2021 daemon.err olsrd[2890]: olsrd exit: main: Bad configuration
Tue Apr 20 02:42:07 2021 user.notice firewall: Reloading firewall due to ifup of wireless0 (wlan0-mesh-2)
~~~

the message of the relevant "link becomes ready" is just before olsrd starts, rather then just after as before, but I assume that just a minimalistic scheduling variance. (indeed i was 3 times this order)

PolynomialDivision, 2021-06-19:
Could you disable olsrd6? Or give me your config?

SvenRoederer, 2021-06-20:
Just using an image based on recent OpenWrt-master. Initially the olsr started up correctly (v4 and v6), but after the night olsrd was also failing with the "Bad configuration" issue. Currently I increased the log-size and will see if I can reproduce this after some time.

After this I can change over to OpenWrt-21.02 based build again and provide the data / tests above.

PolynomialDivision, 2021-06-20:
Could you be a bit more precise? I have that running now for more than 1 week and never had any issues. However, I do not use IPv6 with oslrd. I even have olsrd running on a wdr3600.

SvenRoederer, 2021-06-29:
I did some tests; started with OpenWrt master branch (based on https://github.com/SvenRoederer/freifunk-berlin-firmware/commit/a52246d5434bf2dd602379b132c3c76898816a92):

~~~
OPENWRT_REPO=https://github.com/openwrt/openwrt.git@fe6b9b0588a47e0a6aabc6a51aa676e7e7204174
PACKAGES_PACKAGES_REPO=https://github.com/openwrt/packages.git@238046303d3408c622a2c2c2fc25fbc26cdd20e4
PACKAGES_LUCI_REPO=https://github.com/openwrt/luci.git@d1c4adff93aa5ec0892d0ff7b98d56f21c6e531d
PACKAGES_FREIFUNK_REPO=https://github.com/freifunk/openwrt-packages.git@62948cc1b8d934a709fa5d1104fa230334f979ec
PACKAGES_ROUTING_REPO=https://github.com/openwrt-routing/packages.git@b81e9c8bab72ada2ecb97a9dd36f691aceedda38
~~~

OLSRd started on boot, but the procd-version had several times `olsrd exit: main: Bad configuration` logmessage. So procd tried to start the daemon several times, before it finally succeeded. Attached the syslog, olsr-ucifiles and generated olsr-configs: 
[olsr-procd_on-master.zip](https://github.com/openwrt/routing/files/6736575/olsr-procd_on-master.zip)
Currently rechecking the "olsr fails with bad configuration after running over night" finding of my last comment

I used the same build-env, but only reverted dc0adf7dced5f and installed this rebuild version of olsrd on the same board. After rebooting the system came up without  any error-messages in the log. Attached the same kind of logs / configs.
[olsr-noprocd_on-master.zip](https://github.com/openwrt/routing/files/6736623/olsr-noprocd_on-master.zip)

The procd-implementation on master seems to cause "just" some additional restarts of the daemon, till the links come up. This might be just a cosmetic issue. But based on my previous comment and the behaviour seen on 21.02 there might be more under the hood.

SvenRoederer, 2021-07-01:
findings with OpenWrt-master branch: After some time Olsr began to start/stop with some "Bad Configuration" after some time. I had the system just standing beside with rarely some client connected to the integrated AP and so only low traffic. 
As I was not around when it started, there is only the remaining log I can provide. What I've seen in addition: there was a kernel-error on netflow reported onetime, which might made all start. 
[logread_olsrd_openwrt-master_2days-uptime.txt](https://github.com/openwrt/routing/files/6750504/logread_olsrd_openwrt-master_2days-uptime.txt): 
* 12:09:40 final messages of restarting an olsr6 instance
* 12:09:46 - 12:10:09 olsr was restarted  by external watchdog-script (even it had just changed the smartgateway) - but this all might be unrelated to this issue
* 12:21:01 - 12:27:50: Interface wireless0 (dev wlan0-mesh-2) went down, olsr4 is missing "wireless0 / wlan0-mesh-2" and wireless1 / wlan1-mesh-5"  --> causing "Bad Configuration" message. Same can then be seen for olsr6, but also a message that this crash-loop was already happening 14 times. This goes on for the next 7 minutes.
* 12:40:58 - 12:41:20: another link-down of the wireless interfaces, causing causing a single round of crash-loop

As requested the logs after `/etc/init.d/olsrd6 disable` and reboot: 
[olsr_procd_v4only.zip](https://github.com/openwrt/routing/files/6750621/olsr_procd_v4only.zip) - I don't see a difference to V4 and V6 in parallel.

next-step: extended syslog-buffer to 1MB and will keep it running with V4-only

closed 2021-07-07 by PolynomialDivision

SvenRoederer, 2021-07-07:
> next-step: extended syslog-buffer to 1MB and will keep it running with V4-only

I had the node running the last 6 days, but did not see any WiFi going down, so can't reproduce the situation without putting the node into a shielded box.

Seen that you merged #692 which I'll check soon with openwrt-21.02 branch, as this did not solve the issue seen initially.

PolynomialDivision, 2021-07-08:
> I had the node running the last 6 days, but did not see any WiFi going down, so can't reproduce the situation without putting the node into a shielded box.

Nice! :)

> Seen that you merged #692 which I'll check soon with openwrt-21.02 branch, as this did not solve the issue seen initially.

Noki approved it. Sorry, I'm busy right now. :/ I was not able to reproduce this issue, but noki was and he said it solves it. So I merged since it does not harm (to my knowledge). Feel always free to open PRs. And thanks a lot for testing! :) That is very important.

PolynomialDivision, 2021-07-08:
Feel free to reopen issue again.

PR #692 (2021-07-07): olsrd: fix olsrd starting before network
Should fix #691.
commit 5311c844 (2021-07-07): olsrd: fix olsrd starting before network

Should fix #691.

Sometimes the wifi interface is not ready before olsrd tries to access
it. This leads to warnings in the form of:

daemon.info olsrd: /etc/rc.d/S65olsrd: olsrd_write_interface()
	Warning: Interface 'wireless0' not found, skipped
daemon.notice procd: /etc/rc.d/S65olsrd: olsrd: /etc/rc.d/S65olsrd:
	olsrd_write_interface() Warning: Interface 'wireless0' not found,			skipped

We make use of the "wait_for" procd command that allows us to wait for
the network before starting olsrd.

Tested-by: Noki
PR #700 (2021-07-07): [21.02] olsrd: fix olsrd starting before network
**Urgent fix so cherry-pick also in 21.02**

Should fix #691.

Sometimes the wifi interface is not ready before olsrd tries to access
it. This leads to warnings in the form of:

daemon.info olsrd: /etc/rc.d/S65olsrd: olsrd_write_interface()
	Warning: Interface 'wireless0' not found, skipped
daemon.notice procd: /etc/rc.d/S65olsrd: olsrd: /etc/rc.d/S65olsrd:
	olsrd_write_interface() Warning: Interface 'wireless0' not found,			skipped

We make use of the "wait_for" procd command that allows us to wait for
the network before starting olsrd.
commit 911c0cb4 (2021-07-07): olsrd: fix olsrd starting before network

Should fix #691.

Sometimes the wifi interface is not ready before olsrd tries to access
it. This leads to warnings in the form of:

daemon.info olsrd: /etc/rc.d/S65olsrd: olsrd_write_interface()
	Warning: Interface 'wireless0' not found, skipped
daemon.notice procd: /etc/rc.d/S65olsrd: olsrd: /etc/rc.d/S65olsrd:
	olsrd_write_interface() Warning: Interface 'wireless0' not found,			skipped

We make use of the "wait_for" procd command that allows us to wait for
the network before starting olsrd.

Tested-by: Noki