openwrt/routing #693: babeld:

PolynomialDivision, 2021-06-19:
Maintainer: @me, @kerneis, Baptiste Jonglez
Environment: OpenWrt-21.02-SNAPSHOT (IPQ40xx Fritz!Box 4040)

Description:
Doing a sysupgrade the following boot resulted in a netlink_read error:

```
root@core:~# logread | grep babeld
Sat Jun 19 10:48:05 2021 daemon.err babeld[1907]: netlink_read: recvmsg(): No buffer space available
```

kerneis, 2021-06-19:
@PolynomialDivision is this a reproducible failure or a one-shot error after the first reboot? does it happen if you uninstall and reinstall the package?

@jech in case this rings a bell.

PolynomialDivision, 2021-06-19:
> @PolynomialDivision is this a reproducible failure or a one-shot error after the first reboot? does it happen if you uninstall and reinstall the package?

I will do further investigations. However, this happened on a very important node which I can not flash all the time:
https://hopglass.berlin.freifunk.net/#!v:m;n:sama-core.olsr
I need some time.

jech, 2021-06-19:
Babeld has attempted to install a route, and the kernel replied ENOBUFS.  In principle, babeld should recover cleanly from the failure, it should attempt to install the route again the next time a neighbour announces it.

So it's probably nothing to worry about, although of course it might indicate that the kernel is running out of some resource.

PolynomialDivision, 2021-07-20:
We did some improvements in logging. I saw this error again. Hopefully, we will find a way to reproduce it.

PolynomialDivision, 2021-08-29:
> So it's probably nothing to worry about, although of course it might indicate that the kernel is running out of some resource.

I believe it has something to do with a misbehaving switch of the IPQ40xx SoC. Hopefully, it will be fixed with DSA. I think we can close since it is not babeld related. OLSRD has some issues.:/

closed 2023-03-11 by BKPepe