openwrt/routing #727: cjdns: cjdroute Segmentation fault

gimzmoe, 2021-09-25:
Environment: (put here arch, model, OpenWrt version)
Model: ZyXEL NBG6817
Architecture: ARMv7 Processor rev 0 (v7l)
Firmware Version: OpenWrt 19.07.7 r11306-c4a6851c72 / LuCI openwrt-19.07 branch git-21.044.30835-34e0d65
Kernel Version: 4.14.221
Description:
installed cjdns, cjdroute causes seg fault without any other explaination

```
root@OpenWrt:~# opkg list | grep cjdns
cjdns - 20.2-1

root@OpenWrt:~# cjdroute
Segmentation fault

root@OpenWrt:~# cjdrouteconf get | cjdroute
Segmentation fault
```
Occurs on other tp-link and gl-inet devices too

gimzmoe, 2021-09-25:
trying 21.02

mwarning, 2021-09-25:
@wfleurant

gimzmoe, 2021-09-25:
while 21.02 is quite nice and I'm glad I upgraded, the cjdroute seg fault persists.  No tun0 device

wfleurant, 2021-09-25:
crazy...no? I'll pull and build keeping symbols and see if gdb sheds some light.

wfleurant, 2021-09-27:
can you try 
`gdb ./cjdroute -ex 'set follow-fork-mode child' -ex 'run' -ex 'thread apply all bt' -ex 'quit'`
or use strace? dmesg?

jine, 2021-09-29:
I can confirm we're seeing this issue on GL.iNet B1300, with custom firmware based upon 19.07 (but the cjdns-package is from openwrt master repo, so no modifications there) - CPU: Qualcomm IPQ4028

I can check gdb with those params later this week, as i have another task with one of those units. I remember i also tried 21.02 with the exact same segfault - i found some notes in our internal systems about it i saved:

```
root@xxxxxx-xxxxxx:~# cat /etc/os-release
NAME="xxxxxx"
VERSION="21.02-SNAPSHOT"
ID="openwrt"
ID_LIKE="lede openwrt"
PRETTY_NAME="OpenWrt 21.02-SNAPSHOT"
VERSION_ID="21.02-snapshot"
HOME_URL="https://openwrt.org/"
BUG_URL="https://bugs.openwrt.org/"
SUPPORT_URL="https://forum.openwrt.org/"
BUILD_ID="r0+16120-6bb6901de9"
OPENWRT_BOARD="ipq40xx/generic"
OPENWRT_ARCH="arm_cortex-a7_neon-vfpv4"
OPENWRT_TAINTS="no-all busybox"
OPENWRT_DEVICE_MANUFACTURER="OpenWrt"
OPENWRT_DEVICE_MANUFACTURER_URL="https://openwrt.org/"
OPENWRT_DEVICE_PRODUCT="Generic"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="OpenWrt 21.02-SNAPSHOT r0+16120-6bb6901de9"

root@xxxxxx-xxxxxx:~# cat /etc/openwrt_release
DISTRIB_ID='xxxxxx'
DISTRIB_RELEASE='master'
DISTRIB_REVISION='79dccb0'
DISTRIB_TARGET='ipq40xx/generic'
DISTRIB_ARCH='arm_cortex-a7_neon-vfpv4'
DISTRIB_DESCRIPTION='xxxxxx ********* 20210630_0445'
DISTRIB_TAINTS='no-all busybox'
strace:

root@xxxxx-xxxxxx:~# cjdroute
Segmentation fault
root@xxxxxx-xxxxxx:~# strace -f cjdroute
execve("/usr/sbin/cjdroute", ["cjdroute"], 0xbef52da4 /* 13 vars */) = 0
set_tls(0xb6f5f5d8)                     = 0
set_tid_address(0xb6f6018c)             = 32511
open("/etc/ld-musl-armhf.path", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/lib/libgcc_s.so.1", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
fcntl64(3, F_SETFD, FD_CLOEXEC)         = 0
fstat64(3, {st_mode=S_IFREG|0644, st_size=45059, ...}) = 0
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0000E\0\0004\0\0\0"..., 936) = 936
mmap2(NULL, 114688, PROT_READ|PROT_EXEC, MAP_PRIVATE, 3, 0) = 0xb6ec9000
mmap2(0xb6ee3000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED, 3, 0xa000) = 0xb6ee3000
close(3)                                = 0
mprotect(0xb6ee3000, 4096, PROT_READ)   = 0
--- SIGSEGV {si_signo=SIGSEGV, si_code=SEGV_ACCERR, si_addr=0x4fde1c} ---
+++ killed by SIGSEGV +++
Segmentation fault

```

gimzmoe, 2021-09-29:
```
Reading symbols from /usr/sbin/cjdroute...
(No debugging symbols found in /usr/sbin/cjdroute)
Starting program: /usr/sbin/cjdroute 
warning: Unable to find dynamic linker breakpoint function.
GDB will be unable to debug shared library initializers
and track explicitly loaded dynamic code.

Program received signal SIGSEGV, Segmentation fault.
0xb6fdc598 in ?? ()

Thread 1 (process 32384 "cjdroute"):
#0  0xb6fdc598 in ?? ()
#1  0xb6fdc7c8 in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
A debugging session is active.

	Inferior 1 [process 32384] will be killed.

Quit anyway? (y or n) 
```

jine, 2021-09-30:
Exactly ... 
mprotect(0xb6ee3000, 4096, PROT_READ)   = 0
--- SIGSEGV {si_signo=SIGSEGV, si_code=SEGV_ACCERR, si_addr=0x4fde1c} ---
+++ killed by SIGSEGV +++
Segmentation fault

`As stated in the man page of sigaction, SEGV_ACCERR is a signal code for SIGSEGV that specifies Invalid permissions for mapped object. Contrary to SEGV_MAPERR which means that the address is not mapped to a valid object, SEGV_ACCERR means the address matches an object, but for sure it is neither the good one, nor one the process is allowed to access.`

wfleurant, 2021-10-01:
Thanks for the feedback. I'll find more details soon.

wfleurant, 2021-10-05:
does anyone have more information? broke against musl? I'll try to re-reproduce soon got to run

wfleurant, 2021-10-07:
curses. first task is to fix the builds that are failing now. chances are it fixes this here. the build for your target device wasn't working out.

ancosgrove, 2022-02-25:
Wasn't sure if I should open a new issue and reference this one or re-open. I'm getting the same behavior on a Linksys EA8300. I tried going to the 21-3 snapshot and now get errors on what looks to be whatever libs were linked.

[image]

```
root@Cerberus:/etc# cat os-release
NAME="OpenWrt"
VERSION="21.02.1"
ID="openwrt"
ID_LIKE="lede openwrt"
PRETTY_NAME="OpenWrt 21.02.1"
VERSION_ID="21.02.1"
HOME_URL="https://openwrt.org/"
BUG_URL="https://bugs.openwrt.org/"
SUPPORT_URL="https://forum.openwrt.org/"
BUILD_ID="r16325-88151b8303"
OPENWRT_BOARD="ipq40xx/generic"
OPENWRT_ARCH="arm_cortex-a7_neon-vfpv4"
OPENWRT_TAINTS=""
OPENWRT_DEVICE_MANUFACTURER="OpenWrt"
OPENWRT_DEVICE_MANUFACTURER_URL="https://openwrt.org/"
OPENWRT_DEVICE_PRODUCT="Generic"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="OpenWrt 21.02.1 r16325-88151b8303"
```