openwrt/routing #731: Possible upstream switch to `firewall4` aka `nftables` instead of `iptables`

aparcar, 2021-10-06:
Hi all, especially @openwrt/routing-write,

for the next OpenWrt release `firewall4` is [considered](https://github.com/openwrt/openwrt/pull/4642) as a replacement of the current `iptables` based `firewall` package. While the configuration stays within `/etc/config/firewall`, packages using `iptables` directly may see trouble.

This is a heads up for everyone maintaining such packages but also please post packages here that would be affected so a smother migration is possible.

PolynomialDivision, 2021-10-06:
OLSR is affected.

pmelange, 2021-10-13:
> OLSR is affected.

with a ```git grep -i iptables``` i see that nodogsplash and opennds both require "iptables-mod-ipopt".  What would be the equivalent package in fw4?

In OLSR, it seems like only olsrd/files/olsrd.sh is using iptables and ip6tables in a script.  Is there a syntax change on the command line going from iptables to fw4?

I didn't look at any of the source code of any of the packages, just at the repo itself.

zioproto, 2021-10-17:
@PolynomialDivision is it olsrd core code affected or it is an extension/plugin ? Is it just the [Smart Gateway extension](https://github.com/OLSR/olsrd/blob/v0.9.8/README-Olsr-Extensions#L221) broken ?

aparcar, 2022-01-07:
@openwrt/routing-write  heads up, firewall4 gets into a fine state (thanks to @stintel and @jow- )and I'm planing to make it the default within the next week. For all scripts that require _legacy_ `iptables` please test your apps with `iptables-nft` and report back!

bluewavenet, 2022-01-11:
@aparcar 
Where is `iptables-nft`? (It is probably hidden in clear sight, ie I can't find it :-D )
openNDS works fine with it on "Generic Linux" eg Debian, et al, so should be good here as well at least as an interim solution.

wfleurant, 2022-01-11:
i found something in LuCI. When I Save & Apply a rule under "Traffic Rules" it does not restart the firewall.  
when manually running `/etc/init.d/firewall restart` then all of the rules I enabled/disabled work as expected.  
compiled against your aparcar/openwrt:fw4 (8af05a1) .. tested on rpi4

castillofrancodamian, 2022-01-14:
> @aparcar Where is `iptables-nft`? (It is probably hidden in clear sight, ie I can't find it :-D ) openNDS works fine with it on "Generic Linux" eg Debian, et al, so should be good here as well at least as an interim solution.

The `iptables-nft` package is already there.

aparcar, 2022-01-19:
As discussed at yesterdays meeting I merged the changes. The next release will use `firewall4` as default and all packages incompatible (e.g. using `ipset`) should add a negative dependency. Our considerations are that the default (WiFI home router) setup works fine wir `firewall4` and special cases can always replace `firewall4` with `firewall3`, which should work at least until the upcoming 5.15 Kernels.

PolynomialDivision, 2022-02-28:
Firewall4 fails when adding:
```
config rule                    
        option name 'DSCP olsr'                          
        list proto 'udp'                
        option dest 'freifunk'          
        option dest_port '698'      
        option target 'DSCP'                
        option set_dscp 'EF'      
```

```
/proc/self/fd/0:182:21-27: Error: Byteorder mismatch: expected big endian, got host endian
```

jow-, 2022-03-01:
Please attach the output of `fw4 print`. Looks like an nftables upstream bug though

dave14305, 2022-03-01:
> Please attach the output of `fw4 print`. Looks like an nftables upstream bug though

Like this one?
http://git.netfilter.org/nftables/commit/?id=4788869a3346b5b3f1f6a3cdb12fdf79da961ef9

nftables 1.0.2 was released recently.

BKPepe, 2022-03-06:
> nftables 1.0.2 was released recently.

https://github.com/openwrt/openwrt/pull/9402

PolynomialDivision, 2022-03-06:
Sorry for closing. This was automatically one by merging the PR.

BKPepe, 2022-03-06:
Why this was re-opened? It is not clear to me. OP issue was solved. The issue related to nftables or firewall4 should be moved to somewhere else. :)

PolynomialDivision, 2022-03-06:
Probably @aparcar wants to decide when to close. ;)

aparcar, 2022-03-07:
If OLSR is the only package affected by firewall4 this can be closed.

PR #812 (2022-03-06): olsrd: switch to firewall4
Firewall4 uses nftables instead of iptables. Use iptables-nft for installing smart-gw-rules.

Fixes: #731 ("Certain upstream switch to firewall4 aka nftables instead of iptables")
closed 2022-03-07 by PolynomialDivision