openwrt/routing #755: Update smcroute as actual package has log issues with loopback device

NilsRo, 2021-11-24:
@mwarning To get this watched I created a new issue with a slight different title. ;-)

Followup of https://github.com/openwrt/routing/issues/744

mwarning, 2021-11-24:
I thought this is fixed in smcroute 2.5.3?

NilsRo, 2021-11-24:
There was a buggy build of 2.5.3 which should be fixed with this pull: https://github.com/openwrt/packages/pull/16847

But it was closed before merging with reference to 2.5.4. So 2.5.4 will fix the bug also. I did not find any reference to another pull get the 2.5.3 fixed build into the OpenWRT master so I opened this issue. Perhaps I am wrong...

The 2.5.3 fixed or 2.5.4 is needed. Then everything will be fine.

mwarning, 2021-11-24:
That change was added here https://github.com/troglobit/smcroute/commit/8ce1d117a31e35d97fb955b82edf13514267eaab
and should be in smcroute 2.5.4. So everything should be fine now.

NilsRo, 2021-11-25:
Yes, but it is still not available in OpenWRT and I did not find any pull for 2.5.4.

mwarning, 2021-11-25:
Here is the pull request: https://github.com/openwrt/packages/pull/17188

But the packages for the latest OpenWrt release (21.02) need to be updated.

mwarning, 2022-01-17:
For 21.02: https://github.com/openwrt/packages/pull/17628

closed 2022-01-18 by mwarning