openwrt/routing #814: ` oonf-olsrd2: ` error compiling on openwrt 19.07

mikysal78, 2022-03-05:
Please make sure that the issue subject starts with `<package-name>: `

Maintainer: @HRogge 
Environment: OpenWrt 19.04, TP-Link (3600, 4300, C2600), Linksys wrt3200acm  and gl-mt300n. The system for compiling is Debian 11 (try with 10).

Description:
I using openwrt 19.07 and one year old no broblem, i add new device and recompile for all my device and i have error with olsrd2.
How can i solve?

```
jenkins@jenkins:source $ more build.log |grep oonf-olsrd2
make[3]: Entering directory '/home/jenkins/workspace/Firmware/source/feeds/routing/oonf-olsrd2'
SHELL= flock /home/jenkins/workspace/Firmware/source/tmp/.oonf-olsrd2-0.14.1-1.tar.bz2.flock -c '  	/home/jenkins/workspace/Firmware/source/scripts/download.pl "/home/jenkins/workspace/Firmware/source/dl" "oonf-olsrd2-0.14.1-1.tar.bz2" "18f68d4c92ccda43152f2050b761200681baa452f69342d874e4a651ce883738" "" || (  /home/jenkins/workspace/Firmware/source/scripts/dl_github_archive.py --dl-dir="/home/jenkins/workspace/Firmware/source/dl" --url="https://github.com/OLSR/OONF.git" --version="d86825c0bdf144db706da444be7b0a2e8722b54d" --subdir="oonf-olsrd2-0.14.1-1" --source="oonf-olsrd2-0.14.1-1.tar.bz2" --hash="18f68d4c92ccda43152f2050b761200681baa452f69342d874e4a651ce883738" || ( 	echo "Checking out files from the git repository..."; mkdir -p /home/jenkins/workspace/Firmware/source/tmp/dl && cd /home/jenkins/workspace/Firmware/source/tmp/dl && rm -rf oonf-olsrd2-0.14.1-1 && [ \! -d oonf-olsrd2-0.14.1-1 ] && git clone  https://github.com/OLSR/OONF.git oonf-olsrd2-0.14.1-1 && (cd oonf-olsrd2-0.14.1-1 && git checkout d86825c0bdf144db706da444be7b0a2e8722b54d && git submodule update --init --recursive) && echo "Packing checkout..." && export TAR_TIMESTAMP=`cd oonf-olsrd2-0.14.1-1 && git log -1 --format='\''@%ct'\''` && rm -rf oonf-olsrd2-0.14.1-1/.git && 	tar --numeric-owner --owner=0 --group=0 --mode=a-s --sort=name ${TAR_TIMESTAMP:+--mtime="$TAR_TIMESTAMP"} -c oonf-olsrd2-0.14.1-1 | 	bzip2 -c > /home/jenkins/workspace/Firmware/source/tmp/dl/oonf-olsrd2-0.14.1-1.tar.bz2 && mv /home/jenkins/workspace/Firmware/source/tmp/dl/oonf-olsrd2-0.14.1-1.tar.bz2 /home/jenkins/workspace/Firmware/source/dl/ && rm -rf oonf-olsrd2-0.14.1-1; );  )    '
rm -f /home/jenkins/workspace/Firmware/source/build_dir/target-mips_24kc_musl/oonf-olsrd2-0.14.1-1/.built
touch /home/jenkins/workspace/Firmware/source/build_dir/target-mips_24kc_musl/oonf-olsrd2-0.14.1-1/.built_check
CFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -mips16 -minterlink-mips16 -iremap/home/jenkins/workspace/Firmware/source/build_dir/target-mips_24kc_musl/oonf-olsrd2-0.14.1-1:oonf-olsrd2-0.14.1-1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/home/jenkins/workspace/Firmware/source/staging_dir/target-mips_24kc_musl/usr/include -I/home/jenkins/workspace/Firmware/source/staging_dir/target-mips_24kc_musl/usr/include/libnl-tiny  -I/home/jenkins/workspace/Firmware/source/staging_dir/target-mips_24kc_musl/usr/include -I/home/jenkins/workspace/Firmware/source/staging_dir/target-mips_24kc_musl/include -I/home/jenkins/workspace/Firmware/source/staging_dir/toolchain-mips_24kc_gcc-7.5.0_musl/usr/include -I/home/jenkins/workspace/Firmware/source/staging_dir/toolchain-mips_24kc_gcc-7.5.0_musl/include/fortify -I/home/jenkins/workspace/Firmware/source/staging_dir/toolchain-mips_24kc_gcc-7.5.0_musl/include " CXXFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -mips16 -minterlink-mips16 -iremap/home/jenkins/workspace/Firmware/source/build_dir/target-mips_24kc_musl/oonf-olsrd2-0.14.1-1:oonf-olsrd2-0.14.1-1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/home/jenkins/workspace/Firmware/source/staging_dir/target-mips_24kc_musl/usr/include -I/home/jenkins/workspace/Firmware/source/staging_dir/target-mips_24kc_musl/usr/include/libnl-tiny  -I/home/jenkins/workspace/Firmware/source/staging_dir/target-mips_24kc_musl/usr/include -I/home/jenkins/workspace/Firmware/source/staging_dir/target-mips_24kc_musl/include -I/home/jenkins/workspace/Firmware/source/staging_dir/toolchain-mips_24kc_gcc-7.5.0_musl/usr/include -I/home/jenkins/workspace/Firmware/source/staging_dir/toolchain-mips_24kc_gcc-7.5.0_musl/include/fortify -I/home/jenkins/workspace/Firmware/source/staging_dir/toolchain-mips_24kc_gcc-7.5.0_musl/include " LDFLAGS="-L/home/jenkins/workspace/Firmware/source/staging_dir/target-mips_24kc_musl/usr/lib -L/home/jenkins/workspace/Firmware/source/staging_dir/target-mips_24kc_musl/lib -L/home/jenkins/workspace/Firmware/source/staging_dir/toolchain-mips_24kc_gcc-7.5.0_musl/usr/lib -L/home/jenkins/workspace/Firmware/source/staging_dir/toolchain-mips_24kc_gcc-7.5.0_musl/lib -znow -zrelro " make -j1 -C /home/jenkins/workspace/Firmware/source/build_dir/target-mips_24kc_musl/oonf-olsrd2-0.14.1-1/. AR="mips-openwrt-linux-musl-gcc-ar" AS="mips-openwrt-linux-musl-gcc -c -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -iremap/home/jenkins/workspace/Firmware/source/build_dir/target-mips_24kc_musl/oonf-olsrd2-0.14.1-1:oonf-olsrd2-0.14.1-1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/home/jenkins/workspace/Firmware/source/staging_dir/target-mips_24kc_musl/usr/include -I/home/jenkins/workspace/Firmware/source/staging_dir/target-mips_24kc_musl/usr/include/libnl-tiny" LD=mips-openwrt-linux-musl-ld NM="mips-openwrt-linux-musl-gcc-nm" CC="mips-openwrt-linux-musl-gcc" GCC="mips-openwrt-linux-musl-gcc" CXX="mips-openwrt-linux-musl-g++" RANLIB="mips-openwrt-linux-musl-gcc-ranlib" STRIP=mips-openwrt-linux-musl-strip OBJCOPY=mips-openwrt-linux-musl-objcopy OBJDUMP=mips-openwrt-linux-musl-objdump SIZE=mips-openwrt-linux-musl-size CROSS="mips-openwrt-linux-musl-" ARCH="mips" CMAKE_COMMAND='/home/jenkins/workspace/Firmware/source/staging_dir/host/bin/cmake' CMAKE_DISABLE_cmake_check_build_system=1 olsrd2_static;
make[4]: Entering directory '/home/jenkins/workspace/Firmware/source/build_dir/target-mips_24kc_musl/oonf-olsrd2-0.14.1-1'
make[5]: Entering directory '/home/jenkins/workspace/Firmware/source/build_dir/target-mips_24kc_musl/oonf-olsrd2-0.14.1-1'
make[6]: Entering directory '/home/jenkins/workspace/Firmware/source/build_dir/target-mips_24kc_musl/oonf-olsrd2-0.14.1-1'
make[7]: Entering directory '/home/jenkins/workspace/Firmware/source/build_dir/target-mips_24kc_musl/oonf-olsrd2-0.14.1-1'
make[7]: Leaving directory '/home/jenkins/workspace/Firmware/source/build_dir/target-mips_24kc_musl/oonf-olsrd2-0.14.1-1'
make[7]: Entering directory '/home/jenkins/workspace/Firmware/source/build_dir/target-mips_24kc_musl/oonf-olsrd2-0.14.1-1'
make[7]: Leaving directory '/home/jenkins/workspace/Firmware/source/build_dir/target-mips_24kc_musl/oonf-olsrd2-0.14.1-1'
make[6]: Leaving directory '/home/jenkins/workspace/Firmware/source/build_dir/target-mips_24kc_musl/oonf-olsrd2-0.14.1-1'
make[5]: Leaving directory '/home/jenkins/workspace/Firmware/source/build_dir/target-mips_24kc_musl/oonf-olsrd2-0.14.1-1'
make[4]: Leaving directory '/home/jenkins/workspace/Firmware/source/build_dir/target-mips_24kc_musl/oonf-olsrd2-0.14.1-1'
make[3]: *** [Makefile:91: /home/jenkins/workspace/Firmware/source/build_dir/target-mips_24kc_musl/oonf-olsrd2-0.14.1-1/.built] Error 2
make[3]: Leaving directory '/home/jenkins/workspace/Firmware/source/feeds/routing/oonf-olsrd2'
time: package/feeds/routing/oonf-olsrd2/compile#0.12#0.02#0.13
make[2]: *** [package/Makefile:113: package/feeds/routing/oonf-olsrd2/compile] Error 2

```

PolynomialDivision, 2022-03-05:
@BKPepe

BKPepe, 2022-03-05:
@PolynomialDivision Thanks for mentioning me, but I am sure I didn't break it. In the OpenWrt 19.07 branch, I added only PKG_MIRROR_HASH, which is not related to the issue that has OP.

Looking at [OpenWrt buildbots](https://downloads.openwrt.org/releases/19.07.9/packages/mips_mips32/routing/), I see that oonf-olsrd2 is successfully compiled. Even I checked it with Turris OS, and it is compiled there as well. :-)

I should also mention that GitHub Actions, CI here, compiled it as well.

@mikysal78 If you want to get any help, please, which OpenWrt version and commit hashes in ``feeds.conf`` are you running? You could use **tagged release** 19.07.9 or follow the latest changes in each ``openwrt-19.07`` branch.

Your log, which you provided, is honestly completely useless. There is no indication of error/warning, etc. Nothing that I can see what went wrong.

I suggest running and provide output from:
``make package/oonf-olsrd2/compile V=s``

What could be highly appreciated is the steps to be able to reproduce it. You might want to download and use SDK.

PolynomialDivision, 2022-03-05:
@BKPepe That mention should not mean that I think you broke anything. :)
Thanks for answering. Sorry, I just skipped the issue and thought that you were the last person touching the Makefile, so maybe you have any idea whats wrong.

mikysal78, 2022-03-05:
# feeds.conf
src-git openwisp https://github.com/openwisp/openwisp-config.git;master
src-git packages https://github.com/openwrt/packages.git;openwrt-19.07
src-git luci https://github.com/openwrt/luci.git;openwrt-19.07
src-git targets https://github.com/openwrt/targets.git;master
src-git routing https://github.com/openwrt/routing.git;openwrt-19.07

$ make package/oonf-olsrd2/compile V=s
time: target/linux/prereq#0.22#0.00#0.22
make[1]: Entering directory '/home/jenkins/workspace/Firmware/source'
make[2]: Entering directory '/home/jenkins/workspace/Firmware/source/package/libs/toolchain'
touch /home/jenkins/workspace/Firmware/source/build_dir/target-mips_24kc_musl/toolchain/.prepared_5a0d3c4ce29ceb100185b70e4f5773f4_6664517399ebbbc92a37c5bb081b5c53_check
mkdir -p /home/jenkins/workspace/Firmware/source/build_dir/target-mips_24kc_musl/toolchain
touch /home/jenkins/workspace/Firmware/source/build_dir/target-mips_24kc_musl/toolchain/.prepared_5a0d3c4ce29ceb100185b70e4f5773f4_6664517399ebbbc92a37c5bb081b5c53
rm -f /home/jenkins/workspace/Firmware/source/build_dir/target-mips_24kc_musl/toolchain/.configured_*
rm -f /home/jenkins/workspace/Firmware/source/staging_dir/target-mips_24kc_musl/stamp/.toolchain_installed
(cd /home/jenkins/workspace/Firmware/source/build_dir/target-mips_24kc_musl/toolchain/./; if [ -x ./configure ]; then find /home/jenkins/workspace/Firmware/source/build_dir/target-mips_24kc_musl/toolchain/ -name config.guess | xargs -r chm
od u+w; find /home/jenkins/workspace/Firmware/source/build_dir/target-mips_24kc_musl/toolchain/ -name config.guess | xargs -r -n1 cp --remove-destination /home/jenkins/workspace/Firmware/source/scripts/config.guess; find /home/jenkins/work
space/Firmware/source/build_dir/target-mips_24kc_musl/toolchain/ -name config.sub | xargs -r chmod u+w; find /home/jenkins/workspace/Firmware/source/build_dir/target-mips_24kc_musl/toolchain/ -name config.sub | xargs -r -n1 cp --remove-des
tination /home/jenkins/workspace/Firmware/source/scripts/config.sub; AR="mips-openwrt-linux-musl-gcc-ar" AS="mips-openwrt-linux-musl-gcc -c -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-e
rror=unused-but-set-variable -Wno-error=unused-result -msoft-float -iremap/home/jenkins/workspace/Firmware/source/build_dir/target-mips_24kc_musl/toolchain:toolchain -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -W
l,-z,now -Wl,-z,relro" LD=mips-openwrt-linux-musl-ld NM="mips-openwrt-linux-musl-gcc-nm" CC="mips-openwrt-linux-musl-gcc" GCC="mips-openwrt-linux-musl-gcc" CXX="mips-openwrt-linux-musl-g++" RANLIB="mips-openwrt-linux-musl-gcc-ranlib" STRIP
=mips-openwrt-linux-musl-strip OBJCOPY=mips-openwrt-linux-musl-objcopy OBJDUMP=mips-openwrt-linux-musl-objdump SIZE=mips-openwrt-linux-musl-size CFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour
-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -mips16 -minterlink-mips16 -iremap/home/jenkins/workspace/Firmware/source/build_dir/target-mips_24kc_musl/toolchain:toolchain -Wformat -Werror=format-security 
-fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " CXXFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float
 -mips16 -minterlink-mips16 -iremap/home/jenkins/workspace/Firmware/source/build_dir/target-mips_24kc_musl/toolchain:toolchain -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " CPPFLAGS="-I/ho
me/jenkins/workspace/Firmware/source/staging_dir/target-mips_24kc_musl/usr/include -I/home/jenkins/workspace/Firmware/source/staging_dir/target-mips_24kc_musl/include -I/home/jenkins/workspace/Firmware/source/staging_dir/toolchain-mips_24k
c_gcc-7.5.0_musl/usr/include -I/home/jenkins/workspace/Firmware/source/staging_dir/toolchain-mips_24kc_gcc-7.5.0_musl/include/fortify -I/home/jenkins/workspace/Firmware/source/staging_dir/toolchain-mips_24kc_gcc-7.5.0_musl/include " LDFLAG
S="-L/home/jenkins/workspace/Firmware/source/staging_dir/target-mips_24kc_musl/usr/lib -L/home/jenkins/workspace/Firmware/source/staging_dir/target-mips_24kc_musl/lib -L/home/jenkins/workspace/Firmware/source/staging_dir/toolchain-mips_24k
c_gcc-7.5.0_musl/usr/lib -L/home/jenkins/workspace/Firmware/source/staging_dir/toolchain-mips_24kc_gcc-7.5.0_musl/lib -znow -zrelro "   ./configure --target=mips-openwrt-linux --host=mips-openwrt-linux --build=x86_64-pc-linux-gnu --program
-prefix="" --program-suffix="" --prefix=/usr --exec-prefix=/usr --bindir=/usr/bin --sbindir=/usr/sbin --libexecdir=/usr/lib --sysconfdir=/etc --datadir=/usr/share --localstatedir=/var --mandir=/usr/man --infodir=/usr/info --disable-nls  ; 
fi; )
touch /home/jenkins/workspace/Firmware/source/build_dir/target-mips_24kc_musl/toolchain/.configured_68b329da9893e34099c7d8ad5cb9c940
rm -f /home/jenkins/workspace/Firmware/source/build_dir/target-mips_24kc_musl/toolchain/.built
touch /home/jenkins/workspace/Firmware/source/build_dir/target-mips_24kc_musl/toolchain/.built_check
touch /home/jenkins/workspace/Firmware/source/build_dir/target-mips_24kc_musl/toolchain/.built
rm -rf /home/jenkins/workspace/Firmware/source/build_dir/target-mips_24kc_musl/toolchain/.pkgdir/libc.installed /home/jenkins/workspace/Firmware/source/build_dir/target-mips_24kc_musl/toolchain/.pkgdir/libc
[... cut ...]
make[2]: *** [Makefile:624: /home/jenkins/workspace/Firmware/source/build_dir/target-mips_24kc_musl/toolchain/.pkgdir/libc.installed] Error 1
make[2]: Leaving directory '/home/jenkins/workspace/Firmware/source/package/libs/toolchain'
time: package/libs/toolchain/compile#0.10#0.00#0.10
make[1]: *** [package/Makefile:113: package/libs/toolchain/compile] Error 2
make[1]: Leaving directory '/home/jenkins/workspace/Firmware/source'
make: *** [/home/jenkins/workspace/Firmware/source/include/toplevel.mk:227: package/oonf-olsrd2/compile] Errore 2

Thanks.

PolynomialDivision, 2022-03-05:
Something looks strange with your toolchain? :S
Can you maybe do a clean clone of openwrt and try compiling again?

mikysal78, 2022-03-05:
git clone https://github.com/openwrt/openwrt.git
cd openwrt/
git checkout openwrt-19.07
./scripts/feeds update -a
./scripts/feeds install -a
make defconfig
ionice -c 3 nice -n 20 make package/oonf-olsrd2/compile V=s 2>&1 | tee build.log
more build.log
`
time: target/linux/prereq#0.20#0.01#0.21
make[1]: Entering directory '/home/jenkins/openwrt'
make[2]: Entering directory '/home/jenkins/openwrt/package/libs/toolchain'
touch /home/jenkins/openwrt/build_dir/target-mips_24kc_musl/toolchain/.prepared_bc4b4e45bae91d193ad623afd03af595_6664517399ebbbc92a37c5bb081b5c53_check
mkdir -p /home/jenkins/openwrt/build_dir/target-mips_24kc_musl/toolchain
touch /home/jenkins/openwrt/build_dir/target-mips_24kc_musl/toolchain/.prepared_bc4b4e45bae91d193ad623afd03af595_6664517399ebbbc92a37c5bb081b5c53
rm -f /home/jenkins/openwrt/build_dir/target-mips_24kc_musl/toolchain/.configured_*
rm -f /home/jenkins/openwrt/staging_dir/target-mips_24kc_musl/stamp/.toolchain_installed
(cd /home/jenkins/openwrt/build_dir/target-mips_24kc_musl/toolchain/./; if [ -x ./configure ]; then find /home/jenkins/openwrt/build_dir/target-mips_24kc_musl/toolchain/ -name config.guess | xargs -r chmod u+w; find /home/jenkins/openwrt/b
uild_dir/target-mips_24kc_musl/toolchain/ -name config.guess | xargs -r -n1 cp --remove-destination /home/jenkins/openwrt/scripts/config.guess; find /home/jenkins/openwrt/build_dir/target-mips_24kc_musl/toolchain/ -name config.sub | xargs 
-r chmod u+w; find /home/jenkins/openwrt/build_dir/target-mips_24kc_musl/toolchain/ -name config.sub | xargs -r -n1 cp --remove-destination /home/jenkins/openwrt/scripts/config.sub; AR="mips-openwrt-linux-musl-gcc-ar" AS="mips-openwrt-linu
x-musl-gcc -c -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -iremap/home/jenkins/openwrt/build_dir/target-mips_24kc_mus
l/toolchain:toolchain -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD=mips-openwrt-linux-musl-ld NM="mips-openwrt-linux-musl-gcc-nm" CC="mips-openwrt-linux-musl-gcc" GCC="mips-openwrt-linu
x-musl-gcc" CXX="mips-openwrt-linux-musl-g++" RANLIB="mips-openwrt-linux-musl-gcc-ranlib" STRIP=mips-openwrt-linux-musl-strip OBJCOPY=mips-openwrt-linux-musl-objcopy OBJDUMP=mips-openwrt-linux-musl-objdump SIZE=mips-openwrt-linux-musl-size
 CFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -mips16 -minterlink-mips16 -iremap/home/jenkins/openwrt/build_di
r/target-mips_24kc_musl/toolchain:toolchain -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " CXXFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-c
opts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -mips16 -minterlink-mips16 -iremap/home/jenkins/openwrt/build_dir/target-mips_24kc_musl/toolchain:toolchain -Wformat -Werror=format-security -fstack-protector -D
_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " CPPFLAGS="-I/home/jenkins/openwrt/staging_dir/target-mips_24kc_musl/usr/include -I/home/jenkins/openwrt/staging_dir/target-mips_24kc_musl/include -I/home/jenkins/openwrt/staging_dir/toolchain-mip
s_24kc_gcc-7.5.0_musl/usr/include -I/home/jenkins/openwrt/staging_dir/toolchain-mips_24kc_gcc-7.5.0_musl/include/fortify -I/home/jenkins/openwrt/staging_dir/toolchain-mips_24kc_gcc-7.5.0_musl/include " LDFLAGS="-L/home/jenkins/openwrt/stag
ing_dir/target-mips_24kc_musl/usr/lib -L/home/jenkins/openwrt/staging_dir/target-mips_24kc_musl/lib -L/home/jenkins/openwrt/staging_dir/toolchain-mips_24kc_gcc-7.5.0_musl/usr/lib -L/home/jenkins/openwrt/staging_dir/toolchain-mips_24kc_gcc-
7.5.0_musl/lib -znow -zrelro "   ./configure --target=mips-openwrt-linux --host=mips-openwrt-linux --build=x86_64-pc-linux-gnu --program-prefix="" --program-suffix="" --prefix=/usr --exec-prefix=/usr --bindir=/usr/bin --sbindir=/usr/sbin -
-libexecdir=/usr/lib --sysconfdir=/etc --datadir=/usr/share --localstatedir=/var --mandir=/usr/man --infodir=/usr/info --disable-nls  ; fi; )
touch /home/jenkins/openwrt/build_dir/target-mips_24kc_musl/toolchain/.configured_68b329da9893e34099c7d8ad5cb9c940
rm -f /home/jenkins/openwrt/build_dir/target-mips_24kc_musl/toolchain/.built
touch /home/jenkins/openwrt/build_dir/target-mips_24kc_musl/toolchain/.built_check
touch /home/jenkins/openwrt/build_dir/target-mips_24kc_musl/toolchain/.built
rm -rf /home/jenkins/openwrt/build_dir/target-mips_24kc_musl/toolchain/.pkgdir/libc.installed /home/jenkins/openwrt/build_dir/target-mips_24kc_musl/toolchain/.pkgdir/libc
mkdir -p /home/jenkins/openwrt/build_dir/target-mips_24kc_musl/toolchain/.pkgdir/libc
install -d -m0755 /home/jenkins/openwrt/build_dir/target-mips_24kc_musl/toolchain/.pkgdir/libc/lib /home/jenkins/openwrt/build_dir/target-mips_24kc_musl/toolchain/.pkgdir/libc/usr/bin
cp -fpR /home/jenkins/openwrt/staging_dir/toolchain-mips_24kc_gcc-7.5.0_musl/lib/ld-musl-*.so* /home/jenkins/openwrt/build_dir/target-mips_24kc_musl/toolchain/.pkgdir/libc/lib/
cp: cannot stat '/home/jenkins/openwrt/staging_dir/toolchain-mips_24kc_gcc-7.5.0_musl/lib/ld-musl-*.so*': No such file or directory
make[2]: *** [Makefile:624: /home/jenkins/openwrt/build_dir/target-mips_24kc_musl/toolchain/.pkgdir/libc.installed] Error 1
make[2]: Leaving directory '/home/jenkins/openwrt/package/libs/toolchain'
time: package/libs/toolchain/compile#0.09#0.01#0.10
make[1]: *** [package/Makefile:113: package/libs/toolchain/compile] Error 2
make[1]: Leaving directory '/home/jenkins/openwrt'
make: *** [/home/jenkins/openwrt/include/toplevel.mk:227: package/oonf-olsrd2/compile] Errore 2
`

BKPepe, 2022-03-05:
@mikysal78 You posted here 3 replies and each of them has different formatting. Please stick to the format you used in OP; otherwise, it is a mess, and nobody is going to look at it.

From the steps you provided, it is clear what you are doing wrong. **You can not compile packages, when you don't have compiled toolchains, etc.**

You should follow the documentation first before reporting bugs:
https://openwrt.org/docs/guide-developer/toolchain/use-buildsystem

This is not related to ``oonf-olsrd2`` anyhow. So, this could be closed.

mikysal78, 2022-03-20:
I try... if include only pakage it's ok.
bat if add:
```
CONFIG_OONF_NHDP_AUTOLL4=y
CONFIG_OONF_OLSRV2_LAN_IMPORT=y
```
the error is:
```
make -f CMakeFiles/Makefile2 olsrd2_static
make[5]: Entering directory '/home/jenkins/workspace/Firmware/source/build_dir/target-mips_24kc_musl/oonf-olsrd2-0.14.1-1'
/home/jenkins/workspace/Firmware/source/staging_dir/host/bin/cmake -S/home/jenkins/workspace/Firmware/source/build_dir/target-mips_24kc_musl/oonf-olsrd2-0.14.1-1 -B/home/jenkins/workspace/Firmware/source/build_dir/target-mips_24kc_musl/oonf-olsrd2-0.14.1-1 --check-build-system CMakeFiles/Makefile.cmake 0
/home/jenkins/workspace/Firmware/source/staging_dir/host/bin/cmake -E cmake_progress_start /home/jenkins/workspace/Firmware/source/build_dir/target-mips_24kc_musl/oonf-olsrd2-0.14.1-1/CMakeFiles 1
make -f CMakeFiles/Makefile2 src/olsrd2/CMakeFiles/olsrd2_static.dir/all
make[6]: Entering directory '/home/jenkins/workspace/Firmware/source/build_dir/target-mips_24kc_musl/oonf-olsrd2-0.14.1-1'
make -f src/olsrd2/CMakeFiles/olsrd2_static.dir/build.make src/olsrd2/CMakeFiles/olsrd2_static.dir/depend
make[7]: Entering directory '/home/jenkins/workspace/Firmware/source/build_dir/target-mips_24kc_musl/oonf-olsrd2-0.14.1-1'
cd /home/jenkins/workspace/Firmware/source/build_dir/target-mips_24kc_musl/oonf-olsrd2-0.14.1-1 && /home/jenkins/workspace/Firmware/source/staging_dir/host/bin/cmake -E cmake_depends "Unix Makefiles" /home/jenkins/workspace/Firmware/source/build_dir/target-mips_24kc_musl/oonf-olsrd2-0.14.1-1 /home/jenkins/workspace/Firmware/source/build_dir/target-mips_24kc_musl/oonf-olsrd2-0.14.1-1/src/olsrd2 /home/jenkins/workspace/Firmware/source/build_dir/target-mips_24kc_musl/oonf-olsrd2-0.14.1-1 /home/jenkins/workspace/Firmware/source/build_dir/target-mips_24kc_musl/oonf-olsrd2-0.14.1-1/src/olsrd2 /home/jenkins/workspace/Firmware/source/build_dir/target-mips_24kc_musl/oonf-olsrd2-0.14.1-1/src/olsrd2/CMakeFiles/olsrd2_static.dir/DependInfo.cmake --color=
Dependee "/home/jenkins/workspace/Firmware/source/build_dir/target-mips_24kc_musl/oonf-olsrd2-0.14.1-1/src/olsrd2/CMakeFiles/olsrd2_static.dir/DependInfo.cmake" is newer than depender "/home/jenkins/workspace/Firmware/source/build_dir/target-mips_24kc_musl/oonf-olsrd2-0.14.1-1/src/olsrd2/CMakeFiles/olsrd2_static.dir/depend.internal".
Dependee "/home/jenkins/workspace/Firmware/source/build_dir/target-mips_24kc_musl/oonf-olsrd2-0.14.1-1/src/olsrd2/CMakeFiles/CMakeDirectoryInformation.cmake" is newer than depender "/home/jenkins/workspace/Firmware/source/build_dir/target-mips_24kc_musl/oonf-olsrd2-0.14.1-1/src/olsrd2/CMakeFiles/olsrd2_static.dir/depend.internal".
Scanning dependencies of target olsrd2_static
make[7]: Leaving directory '/home/jenkins/workspace/Firmware/source/build_dir/target-mips_24kc_musl/oonf-olsrd2-0.14.1-1'
make -f src/olsrd2/CMakeFiles/olsrd2_static.dir/build.make src/olsrd2/CMakeFiles/olsrd2_static.dir/build
make[7]: Entering directory '/home/jenkins/workspace/Firmware/source/build_dir/target-mips_24kc_musl/oonf-olsrd2-0.14.1-1'
[100%] Plugin auto_ll4 lan_import is not there, maybe a dependency is missing?
cd /home/jenkins/workspace/Firmware/source/build_dir/target-mips_24kc_musl/oonf-olsrd2-0.14.1-1/src/olsrd2 && false
make[7]: *** [src/olsrd2/CMakeFiles/olsrd2_static.dir/build.make:58: src/olsrd2/CMakeFiles/olsrd2_static] Error 1
make[7]: Leaving directory '/home/jenkins/workspace/Firmware/source/build_dir/target-mips_24kc_musl/oonf-olsrd2-0.14.1-1'
make[6]: *** [CMakeFiles/Makefile2:6015: src/olsrd2/CMakeFiles/olsrd2_static.dir/all] Error 2
make[6]: Leaving directory '/home/jenkins/workspace/Firmware/source/build_dir/target-mips_24kc_musl/oonf-olsrd2-0.14.1-1'
make[5]: *** [CMakeFiles/Makefile2:6022: src/olsrd2/CMakeFiles/olsrd2_static.dir/rule] Error 2
make[5]: Leaving directory '/home/jenkins/workspace/Firmware/source/build_dir/target-mips_24kc_musl/oonf-olsrd2-0.14.1-1'
make[4]: *** [Makefile:2283: olsrd2_static] Error 2
make[4]: Leaving directory '/home/jenkins/workspace/Firmware/source/build_dir/target-mips_24kc_musl/oonf-olsrd2-0.14.1-1'
make[3]: *** [Makefile:91: /home/jenkins/workspace/Firmware/source/build_dir/target-mips_24kc_musl/oonf-olsrd2-0.14.1-1/.built] Error 2
make[3]: Leaving directory '/home/jenkins/workspace/Firmware/source/feeds/routing/oonf-olsrd2'
time: package/feeds/routing/oonf-olsrd2/compile#1.51#0.40#1.75
make[2]: *** [package/Makefile:113: package/feeds/routing/oonf-olsrd2/compile] Error 2
make[2]: Leaving directory '/home/jenkins/workspace/Firmware/source'
make[1]: *** [package/Makefile:107: /home/jenkins/workspace/Firmware/source/staging_dir/target-mips_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/jenkins/workspace/Firmware/source'
make: *** [/home/jenkins/workspace/Firmware/source/include/toplevel.mk:227: world] Errore 2
jenkins@jenkins:source $ 
```

closed 2022-03-24 by mikysal78