openwrt/routing #817: olsr: error recursive dependency detected!

Alexey71, 2022-03-07:
```
tmp/.config-package.in:103488:error: recursive dependency detected!
tmp/.config-package.in:103488:  symbol PACKAGE_shorewall6 depends on PACKAGE_shorewall6
For a resolution refer to Documentation/kbuild/kconfig-language.rst
subsection "Kconfig recursive dependency limitations"

tmp/.config-package.in:103524:error: recursive dependency detected!
tmp/.config-package.in:103524:  symbol PACKAGE_shorewall6-lite depends on PACKAGE_shorewall6-lite
For a resolution refer to Documentation/kbuild/kconfig-language.rst
subsection "Kconfig recursive dependency limitations"

tmp/.config-package.in:109189:error: recursive dependency detected!
tmp/.config-package.in:109189:  symbol PACKAGE_olsrd depends on PACKAGE_olsrd
For a resolution refer to Documentation/kbuild/kconfig-language.rst
subsection "Kconfig recursive dependency limitations"

tmp/.config-package.in:61438:error: recursive dependency detected!
tmp/.config-package.in:61438:   symbol PACKAGE_luci-app-olsr depends on PACKAGE_luci-app-olsr
For a resolution refer to Documentation/kbuild/kconfig-language.rst
subsection "Kconfig recursive dependency limitations"

tmp/.config-package.in:61455:error: recursive dependency detected!
tmp/.config-package.in:61455:   symbol PACKAGE_luci-app-olsr-services depends on PACKAGE_luci-app-olsr-services
For a resolution refer to Documentation/kbuild/kconfig-language.rst
subsection "Kconfig recursive dependency limitations"

tmp/.config-package.in:61471:error: recursive dependency detected!
tmp/.config-package.in:61471:   symbol PACKAGE_luci-app-olsr-viz depends on PACKAGE_luci-app-olsr-viz
For a resolution refer to Documentation/kbuild/kconfig-language.rst
subsection "Kconfig recursive dependency limitations"
```

Alexey71, 2022-03-09:
@PolynomialDivision

BKPepe, 2022-03-09:
shorewall recursive dependency is already reported here: https://github.com/openwrt/packages/issues/17964

closed 2022-03-14 by Alexey71

BKPepe, 2022-03-14:
Should by fixed by this: https://github.com/openwrt/routing/commit/1cd72bff838ef6de489f8fda28487bd5a4e83c82