openwrt/routing #827: opennds: error: recursive dependency detected!

Alexey71, 2022-03-25:
After update https://github.com/openwrt/routing/commit/d078190b846d518f4a98e976a01525caf7edc6bb
```
tmp/.config-package.in:101924:error: recursive dependency detected!
tmp/.config-package.in:101924:  symbol PACKAGE_iptables-legacy is selected by PACKAGE_iptables-nft
tmp/.config-package.in:102731:  symbol PACKAGE_iptables-nft is selected by PACKAGE_opennds
tmp/.config-package.in:100954:  symbol PACKAGE_opennds depends on PACKAGE_iptables-legacy
For a resolution refer to Documentation/kbuild/kconfig-language.rst
subsection "Kconfig recursive dependency limitations"
```

Alexey71, 2022-03-25:
@mwarning @bluewavenet

bluewavenet, 2022-03-25:
@Alexey71 
Where are you seeing this problem?
Extensive testing here does not give any problems.
Are you installing v9.7.0-2 on snapshot?
Have you installed iptables(legacy) before installing openNDS?

Alexey71, 2022-03-25:
@bluewavenet 
Yes Snapshot

```
alexey@DESKTOP:~$ cd ~/openwrt && rm -rf tmp && git pull origin master && ./scripts/feeds update -a && ./scripts/feeds install -a && make menuconfig
From https://github.com/openwrt/openwrt
 * branch                  master     -> FETCH_HEAD
Already up to date.
Updating feed 'packages' from 'https://git.openwrt.org/feed/packages.git' ...
Already up to date.
Updating feed 'luci' from 'https://git.openwrt.org/project/luci.git' ...
Already up to date.
Updating feed 'routing' from 'https://git.openwrt.org/feed/routing.git' ...
Already up to date.
Updating feed 'telephony' from 'https://git.openwrt.org/feed/telephony.git' ...
Already up to date.
Create index file './feeds/packages.index'
Create index file './feeds/luci.index'
Create index file './feeds/routing.index'
Create index file './feeds/telephony.index'
Installing all packages from feed packages.
Installing all packages from feed luci.
Installing all packages from feed routing.
Installing all packages from feed telephony.
tmp/.config-package.in:101943:error: recursive dependency detected!
tmp/.config-package.in:101943:  symbol PACKAGE_iptables-legacy is selected by PACKAGE_iptables-nft
tmp/.config-package.in:102750:  symbol PACKAGE_iptables-nft is selected by PACKAGE_opennds
tmp/.config-package.in:100973:  symbol PACKAGE_opennds depends on PACKAGE_iptables-legacy
For a resolution refer to Documentation/kbuild/kconfig-language.rst
subsection "Kconfig recursive dependency limitations"

*** End of the configuration.
*** Execute 'make' to start the build or try 'make help'.

alexey@DESKTOP:~/openwrt$
```

bluewavenet, 2022-03-25:
@Alexey71 
I regularly build using SDK and do not have any issues. Also the OpenWrt build bots are not having problems.
Is this a problem with your setup? 
Maybe you need to do `make clean` or `make dirclean` before updating the feeds... not really my area of expertise.

Alexey71, 2022-03-25:
@bluewavenet 
I have downloaded the source code. I didn't change any settings in menuconfig. After exiting menuconfig the error reappears

I see not only I have a mistake
https://github.com/openwrt/openwrt/issues/9407#issuecomment-1073560281

bluewavenet, 2022-03-25:
@Alexey71 
See my comments on https://github.com/openwrt/openwrt/issues/9407#issuecomment-1078822824

Alexey71, 2022-03-26:
@bluewavenet 
Can try https://github.com/Alexey71/routing/commit/8399a927961fd006c719f11ba3ffe5e2f98341b3
No more error

bluewavenet, 2022-03-26:
@Alexey71 
It is this line that is the problem:
`		select PACKAGE_iptables-legacy if PACKAGE_iptables-nft<PACKAGE_opennds`

It is incorrect and is not present in the config-package.in created by the sdk

Alexey71, 2022-03-26:
@bluewavenet 
Ok. What if just delete`iptables-legacy`
in 33 line
`CONFLICTS:=nodogsplash nodogsplash2 iptables-legacy`
Error is gone

bluewavenet, 2022-03-26:
@Alexey71 
See:
https://github.com/openwrt/openwrt/issues/9540#issuecomment-1079649881

bluewavenet, 2022-03-26:
@Alexey71 

>  Ok. What if just delete iptables-legacy

Yes that will work, but what happens if someone already has iptables-legacy installed?
They will end up with both iptables-legacy and iptables-nft with unpredictable results.

Alexey71, 2022-03-26:
@bluewavenet
As far as I know, Snaphot (branch - master) uses nft by default.
Let's wait to see if anyone else has an answer.

bluewavenet, 2022-03-26:
@Alexey71 

>  As far as I know, Snaphot (branch - master) uses nft by default.

Yes it does. In addition, it does not have iptables installed by default.

The problem is that the bug in the iptables makefile results in the installation of iptables-legacy and allows concurrent installation of iptables-nft - which is really bad.

It seems that build-bots and sdk take the last definition for PROVIDES, but the tool chain takes the first value.....

Alexey71, 2022-04-13:
fixed https://github.com/openwrt/routing/commit/8574cb411a1437ad09f047c0e683a126d8ccdd95

closed 2022-04-13 by Alexey71

bluewavenet, 2022-04-13:
@Alexey71 
Be aware this fix, with the current iptables issue, will dramatically compromise the performance of openNDS when quotas are enabled if iptables-legacy is present. It also can result in conflicts with numerous other packages that also use iptables.
You will have to exercise a high level of caution when installing openNDS.