openwrt/routing #850: batman-adv: invalid operands to binary && (have 'int' and 'atomic_t')

zhanhb, 2022-04-17:
branch: master
The source of batman-adv-2022.0 and kernel 5.10.111 are not compatible.
The type of mc_forwarding has change to atomic_t from __s32.
The changes are also applied to 5.4.189, 5.10.111, 5.15.34, 5.16.20, 5.17.3.

See also: https://github.com/torvalds/linux/commit/145c7a793838add5e004e7d49a67654dc7eba147
```txt
/builder/shared-workdir/build/build_dir/target-mips_24kc_musl/linux-ath79_nand/batman-adv-2022.0/net/batman-adv/multicast.c:137:21: error: invalid operands to binary && (have 'int' and 'atomic_t') 
  137 |         if (in6_dev && in6_dev->cnf.mc_forwarding)
      |             ~~~~~~~ ^~ ~~~~~~~~~~~~~~~~~~~~~~~~~~
      |             |                      |   
      |             int                    atomic_t
```

zhanhb, 2022-04-17:
suggest solution:
1. just ignore the old kernel versions.
2. move the package batman-adv to feed base.
3. create a patch like as following
```patch
--- a/net/batman-adv/multicast.c
+++ b/net/batman-adv/multicast.c
@@ -134,7 +134,11 @@ static u8 batadv_mcast_mla_rtr_flags_softif_get_ipv6(struct net_device *dev)
 {
        struct inet6_dev *in6_dev = __in6_dev_get(dev);
 
+#if KERNEL_VERSION(5, 4, 189) <= LINUX_VERSION_CODE && LINUX_VERSION_CODE < KERNEL_VERSION(5, 5, 0) || KERNEL_VERSION(5, 10, 111) <= LINUX_VERSION_CODE && LINUX_VERSION_CODE < KERNEL_VERSION(5, 11, 0) || KERNEL_VERSION(5, 15, 34) <= LINUX_VERSION_CODE && LINUX_VERSION_CODE < KERNEL_VERSION(5, 16, 0) || KERNEL_VERSION(5, 16, 20) <= LINUX_VERSION_CODE && LINUX_VERSION_CODE < KERNEL_VERSION(5, 17, 0) || LINUX_VERSION_CODE >= KERNEL_VERSION(5, 17, 3)
+       if (in6_dev && atomic_read(&in6_dev->cnf.mc_forwarding))
+#else
        if (in6_dev && in6_dev->cnf.mc_forwarding)
+#endif
                return BATADV_NO_FLAGS;
        else
```

Boos4721, 2022-04-17:
GB, try pr it.

commit 73195d63 (2022-04-18): batman-adv: compat: Add atomic mc_forwarding support for stable kernels

Eric Dumazet changed the mc_forwarding in commit 145c7a793838 ("ipv6: make
mc_forwarding atomic") the type of mc_forwarding from __s32 to an atomic_t.
This patch was then ported to various stable kernels.

This code change caused a FTBFS when in batman-adv when
CONFIG_BATMAN_ADV_MCAST was activated. To work around this problem, provide
a version for kernels with __s32 mc_forwarding and a version for kernel
with atomic_t mc_forwarding.

Fixes: #850
Reported-by: Huangbin Zhan <zhanhb88@gmail.com>
PR #853 (2022-04-18): openwrt-21.02: batman-adv: compat: Add atomic mc_forwarding support for stable kernels
Maintainer: @simonwunderlich 
Compile tested: ipq40xx
Run tested: -

Description:

Eric Dumazet changed the mc_forwarding in commit 145c7a793838 ("ipv6: make mc_forwarding atomic") the type of mc_forwarding from __s32 to an atomic_t. This patch was then ported to various stable kernels.

This code change caused a FTBFS when in batman-adv when `CONFIG_BATMAN_ADV_MCAST` was activated. To work around this problem, provide a version for kernels with __s32 mc_forwarding and a version for kernel with atomic_t mc_f [... cut]
commit b8461cc2 (2022-04-18): batman-adv: compat: Add atomic mc_forwarding support for stable kernels

Eric Dumazet changed the mc_forwarding in commit 145c7a793838 ("ipv6: make
mc_forwarding atomic") the type of mc_forwarding from __s32 to an atomic_t.
This patch was then ported to various stable kernels.

This code change caused a FTBFS when in batman-adv when
CONFIG_BATMAN_ADV_MCAST was activated. To work around this problem, provide
a version for kernels with __s32 mc_forwarding and a version for kernel
with atomic_t mc_forwarding.

Fixes: #850
Reported-by: Huangbin Zhan <zhanhb88@gmail.com>
PR #854 (2022-04-18): openwrt-22.03: batman-adv: compat: Add atomic mc_forwarding support for stable kernels
Maintainer: @simonwunderlich 
Compile tested: ipq40xx
Run tested: -

Description:

Eric Dumazet changed the mc_forwarding in commit 145c7a793838 ("ipv6: make mc_forwarding atomic") the type of mc_forwarding from __s32 to an atomic_t. This patch was then ported to various stable kernels.

This code change caused a FTBFS when in batman-adv when `CONFIG_BATMAN_ADV_MCAST` was activated. To work around this problem, provide a version for kernels with __s32 mc_forwarding and a version for kernel with atomic_t mc_f [... cut]
closed 2022-04-18 by simonwunderlich