openwrt/routing #855: batman-adv-2022.0  compat-hacks.h:32:53: error: 'struct br_ip' has no member named 'dst'

hpclass, 2022-04-19:
Environment: (put here arch, model, OpenWrt version)
ARCH:MIPISEL
Model:Xiaomi-Mi-CR6608
OpenWrt R22.4.1 / LuCI Master (git-22.107.56462-5895936)
kernel  5.4.188
**Description:**

```
When I select easy-mesh  with build-in and then show me this error ,and I track compat-hacks.h find then "struct br_ip_list"
it show me this :
./build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/batman-adv-2022.0/compat-include/linux/if_bridge.h

struct batadv_br_ip {
        union {
                __be32  ip4;
#if IS_ENABLED(CONFIG_IPV6)
                struct in6_addr ip6;
#endif
        } dst;
        __be16          proto;
        __u16           vid;
};

struct batadv_br_ip_list {
        struct list_head list;
        struct batadv_br_ip addr;
};

So I can see menber **dst** is exitsing on "struct batadv_br_ip" ,Why it does not working ?
IPV6 is set ENABLE.
```
```
**here is the Log:**

patching file net/batman-adv/multicast.c
touch /home/hp/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/batman-adv-2022.0/.prepared_87b837302af8ae00465c745340d72f28_6664517399ebbbc92a37c5bb081b5c53
rm -f /home/hp/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/batman-adv-2022.0/.configured_*
rm -f /home/hp/lede/staging_dir/target-mipsel_24kc_musl/stamp/.batman-adv_installed
(cd /home/hp/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/batman-adv-2022.0/./; if [ -x ./configure ]; then find /home/hp/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/batman-adv-2022.0/ -name config.guess | xargs -r chmod u+w; find /home/hp/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/batman-adv-2022.0/ -name config.guess | xargs -r -n1 cp --remove-destination /home/hp/lede/scripts/config.guess; find /home/hp/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/batman-adv-2022.0/ -name config.sub | xargs -r chmod u+w; find /home/hp/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/batman-adv-2022.0/ -name config.sub | xargs -r -n1 cp --remove-destination /home/hp/lede/scripts/config.sub; AR="mipsel-openwrt-linux-musl-gcc-ar" AS="mipsel-openwrt-linux-musl-gcc -c -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -fmacro-prefix-map=/home/hp/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/batman-adv-2022.0=batman-adv-2022.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD=mipsel-openwrt-linux-musl-ld NM="mipsel-openwrt-linux-musl-gcc-nm" CC="mipsel-openwrt-linux-musl-gcc" GCC="mipsel-openwrt-linux-musl-gcc" CXX="mipsel-openwrt-linux-musl-g++" RANLIB="mipsel-openwrt-linux-musl-gcc-ranlib" STRIP=mipsel-openwrt-linux-musl-strip OBJCOPY=mipsel-openwrt-linux-musl-objcopy OBJDUMP=mipsel-openwrt-linux-musl-objdump SIZE=mipsel-openwrt-linux-musl-size CFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -mips16 -minterlink-mips16 -fmacro-prefix-map=/home/hp/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/batman-adv-2022.0=batman-adv-2022.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " CXXFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -mips16 -minterlink-mips16 -fmacro-prefix-map=/home/hp/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/batman-adv-2022.0=batman-adv-2022.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " CPPFLAGS="-I/home/hp/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/usr/include -I/home/hp/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/include/fortify -I/home/hp/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/include " LDFLAGS="-L/home/hp/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/usr/lib -L/home/hp/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/lib -znow -zrelro "   ./configure --target=mipsel-openwrt-linux --host=mipsel-openwrt-linux --build=x86_64-pc-linux-gnu --program-prefix="" --program-suffix="" --prefix=/usr --exec-prefix=/usr --bindir=/usr/bin --sbindir=/usr/sbin --libexecdir=/usr/lib --sysconfdir=/etc --datadir=/usr/share --localstatedir=/var --mandir=/usr/man --infodir=/usr/info --disable-nls  ; fi; )
touch /home/hp/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/batman-adv-2022.0/.configured_68b329da9893e34099c7d8ad5cb9c940
rm -f /home/hp/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/batman-adv-2022.0/.built
touch /home/hp/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/batman-adv-2022.0/.built_check
make  -C "/home/hp/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.188" KCFLAGS="-fmacro-prefix-map=/home/hp/lede/build_dir/target-mipsel_24kc_musl=target-mipsel_24kc_musl" HOSTCFLAGS="-O2 -I/home/hp/lede/staging_dir/host/include -I/home/hp/lede/staging_dir/hostpkg/include -I/home/hp/lede/staging_dir/target-mipsel_24kc_musl/host/include -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="mipsel-openwrt-linux-musl-" ARCH="mips" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Tue Apr 19 10:50:47 2022" KBUILD_BUILD_VERSION="0" HOST_LOADLIBES="-L/home/hp/lede/staging_dir/host/lib" KBUILD_HOSTLDLIBS="-L/home/hp/lede/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls= KBUILD_EXTRA_SYMBOLS="/home/hp/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/symvers/netatop.symvers /home/hp/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/symvers/cryptodev-linux.symvers /home/hp/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/symvers/gpio-button-hotplug.symvers /home/hp/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/symvers/mtk-eip93.symvers /home/hp/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/symvers/mac80211.symvers /home/hp/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/symvers/mt76.symvers" KERNELRELEASE=5.4.188 M="/home/hp/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/batman-adv-2022.0/net/batman-adv" CONFIG_BATMAN_ADV=m CONFIG_BATMAN_ADV_DEBUG=n CONFIG_BATMAN_ADV_BLA=y CONFIG_BATMAN_ADV_DAT=y CONFIG_BATMAN_ADV_MCAST=y CONFIG_BATMAN_ADV_NC=n CONFIG_BATMAN_ADV_BATMAN_V=y CONFIG_BATMAN_ADV_TRACING=n  EXTRA_CFLAGS=" -DCONFIG_BATMAN_ADV=1  -DCONFIG_BATMAN_ADV_BLA=1  -DCONFIG_BATMAN_ADV_DAT=1  -DCONFIG_BATMAN_ADV_MCAST=1  -DCONFIG_BATMAN_ADV_BATMAN_V=1 " NOSTDINC_FLAGS="-nostdinc  -isystem /home/hp/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/lib/gcc/mipsel-openwrt-linux-musl/8.4.0/include -I/home/hp/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/batman-adv-2022.0/net/batman-adv -I/home/hp/lede/staging_dir/target-mipsel_24kc_musl/usr/include/mac80211-backport -I/home/hp/lede/staging_dir/target-mipsel_24kc_musl/usr/include/mac80211-backport/uapi -I/home/hp/lede/staging_dir/target-mipsel_24kc_musl/usr/include/mac80211 -I/home/hp/lede/staging_dir/target-mipsel_24kc_musl/usr/include/mac80211/uapi -I/home/hp/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/batman-adv-2022.0/include/ -include backport/autoconf.h -include backport/backport.h -include /home/hp/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/batman-adv-2022.0/compat-hacks.h -DBATADV_SOURCE_VERSION=\\\"2022.0-openwrt-4\\\"" modules
make[3]: Entering directory '/home/hp/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.188'
  CC [M]  /home/hp/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/batman-adv-2022.0/net/batman-adv/bat_algo.o
In file included from <command-line>:
/home/hp/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/batman-adv-2022.0/compat-hacks.h: In function 'br_multicast_has_router_adjacent':
/home/hp/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/batman-adv-2022.0/compat-hacks.h:32:53: error: 'struct br_ip' has no member named 'dst'
       ipv6_addr_is_ll_all_routers(&br_ip_entry->addr.dst.ip6))
                                                     ^
make[4]: *** [scripts/Makefile.build:262: /home/hp/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/batman-adv-2022.0/net/batman-adv/bat_algo.o] Error 1
[... cut ...]
make[2]: *** [Makefile:95: /home/hp/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/batman-adv-2022.0/.built] Error 2
make[2]: Leaving directory '/home/hp/lede/feeds/routing/batman-adv'
time: package/feeds/routing/batman-adv/compile#0.62#1.96#5.06
    ERROR: package/feeds/routing/batman-adv failed to build.
make[1]: *** [package/Makefile:116: package/feeds/routing/batman-adv/compile] Error 1
make[1]: Leaving directory '/home/hp/lede'
make: *** [/home/hp/lede/include/toplevel.mk:230: package/batman-adv/compile] Error 2

jin3014test2, 2022-04-20:
我也遇到这错误了，make一下

lingandken, 2022-04-20:
我也遇到了。。求教

hpclass, 2022-04-20:
Sorry I found the wrong file, the correct include file should be: /home/hp/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.188/include/linux/if_bridge.h

And the "struct br_ip" like this :

struct br_ip {
        union {
                __be32  ip4;
#if IS_ENABLED(CONFIG_IPV6)
                struct in6_addr ip6;
#endif
        } u;
        __be16          proto;
        __u16           vid;
};

struct br_ip_list {
        struct list_head list;
        struct br_ip addr;
};
So it will not work.I will try to Fix it later

hpclass, 2022-04-20:
这是个天坑，，等Dalao弄吧，我追不下去了。以下是我的笔记：
/home/hp/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.188/include/linux/if_bridge.h
在以上路径添加uniou{
。。。
} dst ;
cp /home/hp/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/batman-adv-2022.0/compat-include/linux/minmax.h /home/hp/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.188/include/linux/

报错：
/home/hp/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/batman-adv-2022.0/net/batman-adv/netlink.c:1360:36: error: array type has incomplete element type 'struct genl_small_ops'
 static const struct genl_small_ops batadv_netlink_ops[] = {

kernel 丢失结构体“struct genl_small_ops“
kernel 结构体“ struct genl_family“缺失成员n_small_ops

因为不理解程序的原理，不能乱改，故就此停手。
直接原因是因为Package与kernel 版本不匹配造成的。

zhengmz, 2022-04-21:
```shell
rm -fr build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/batman-adv-2022.0
rm -fr feeds/routing/batman-adv
svn co https://github.com/openwrt/routing/branches/openwrt-22.03/batman-adv/ feeds/routing/batman-adv
```

感觉 master 分支还没合并 22.03 分支的内容

ecsv, 2022-04-21:
Sorry, I can only support the official builds. And Linux 5.4.188 is only used on OpenWrt 21.02. And it builds on fine OpenWrt 21.02 (as you can see in the v21.02.3 release). But you might have noticed that OpenWrt 21.02 is not shipping batman-adv 2022.0. This version is only used by OpenWrt 22.03 and master.

So I don't have any idea what you guys are doing but it is nothing which is officially supported. If I am wrong, please tell me exactly what official OpenWrt version you are using with which which config. I can then try to reproduce it and fix it. And no, R22.4.1 / LuCI Master (git-22.107.56462-5895936) is nothing official from OpenWrt.

hpclass, 2022-04-22:
yep I use this source : https://github.com/coolsnowwolf/lede ；(master  c015f9b54 kernel: bump 5.10 to 5.10.111 (#9259)) So I will change source to : https://github.com/openwrt/openwrt ; and try again.

hpclass, 2022-04-22:
Thanks all guys, I am a newer to OpenWRT, and now I understand the relationship between the OenWRT branch, and the downstream branch of this problem has proposed a solution.
https://github.com/coolsnowwolf/lede/issues/9291

closed 2022-04-22 by hpclass