openwrt/routing #882: ERROR: package/feeds/routing/cjdns failed to build.

zhangguanzhang, 2022-06-11:
`<package-name>: ` cjdns

Maintainer: @William Fleurant <meshnet@protonmail.com>
Environment: on build

Description:
<details>
  <summary>build log</summary>

```
make[3]: Entering directory '/workdir/openwrt/feeds/routing/cjdns'
rm -f /workdir/openwrt/build_dir/target-aarch64_generic_musl/cjdns-cjdns-v21/.built
touch /workdir/openwrt/build_dir/target-aarch64_generic_musl/cjdns-cjdns-v21/.built_check
install -d -m0755 /workdir/openwrt/build_dir/target-aarch64_generic_musl/cjdns-cjdns-v21/tmp
(cd /workdir/openwrt/build_dir/target-aarch64_generic_musl/cjdns-cjdns-v21 && CROSS="true" CC="ccache_cc" AR="aarch64-openwrt-linux-musl-gcc-ar" RANLIB="aarch64-openwrt-linux-musl-gcc-ranlib" CFLAGS="-Os -pipe -mcpu=generic -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/workdir/openwrt/build_dir/target-aarch64_generic_musl/cjdns-cjdns-v21=cjdns-cjdns-v21 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -U_FORTIFY_SOURCE -Wno-error=array-bounds -Wno-error=stringop-overflow -Wno-error=stringop-overread" LDFLAGS="-L/workdir/openwrt/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/usr/lib -L/workdir/openwrt/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/lib -znow -zrelro" SYSTEM="linux" TARGET_ARCH=""aarch64"" SSP_SUPPORT="y" GYP_ADDITIONAL_ARGS="-f make-linux" CJDNS_BUILD_TMPDIR="/workdir/openwrt/build_dir/target-aarch64_generic_musl/cjdns-cjdns-v21/tmp" CJDNS_RELEASE_VERSION= exec ./do)
timeout is 600000
Initialize 211ms
{"isLLVM":false,"isClang":false,"isGCC":true,"version":"11.3.0"}
testing python python3.7
error starting python Error: spawn python3.7 ENOENT
testing python python3
sys.version_info(major=3, minor=10, micro=4, releaselevel='final', serial=0)
Build Libuv
Traceback (most recent call last):
  File "/workdir/openwrt/build_dir/target-aarch64_generic_musl/cjdns-cjdns-v21/build_linux/dependencies/libuv/./gyp_uv.py", line 14, in <module>
    import gyp
  File "/workdir/openwrt/build_dir/target-aarch64_generic_musl/cjdns-cjdns-v21/build_linux/dependencies/libuv/build/gyp/pylib/gyp/__init__.py", line 10, in <module>
    import gyp.input
  File "/workdir/openwrt/build_dir/target-aarch64_generic_musl/cjdns-cjdns-v21/build_linux/dependencies/libuv/build/gyp/pylib/gyp/input.py", line 8, in <module>
    import gyp.common
  File "/workdir/openwrt/build_dir/target-aarch64_generic_musl/cjdns-cjdns-v21/build_linux/dependencies/libuv/build/gyp/pylib/gyp/common.py", line 497, in <module>
    class OrderedSet(collections.MutableSet):
AttributeError: module 'collections' has no attribute 'MutableSet'
make[4]: *** out: No such file or directory.  Stop.
Configure 94ms
Scan for out of date files 2ms
Cross compiling. Building, but not running tests.
Compiler supports link time optimization
YES
cc1:note: someone does not honour COPTS correctly, passed 18 times
Building C object contrib/c/publictoip6.c complete
cc1:note: someone does not honour COPTS correctly, passed 18 times
Building C object contrib/c/privatetopublic.c complete
cc1:note: someone does not honour COPTS correctly, passed 18 times
Building C object client/cjdroute2.c complete
cc1:note: someone does not honour COPTS correctly, passed 18 times
Building C object contrib/c/makekeys.c complete
cc1:note: someone does not honour COPTS correctly, passed 18 times
Building C object contrib/c/sybilsim.c complete
cc1:note: someone does not honour COPTS correctly, passed 18 times
Building C object contrib/c/mkpasswd.c complete
cc1:note: someone does not honour COPTS correctly, passed 18 times
Building C object crypto/random/randombytes.c complete
cc1:note: someone does not honour COPTS correctly, passed 18 times
Building C object test/testcjdroute.c complete
Searching for SECCOMP
cc1:note: someone does not honour COPTS correctly, passed 18 times
Building C object crypto/Key.c complete
cc1:note: someone does not honour COPTS correctly, passed 18 times
Building C object util/AddrTools.c complete
cc1:note: someone does not honour COPTS correctly, passed 18 times
Building C object crypto/random/Random.c complete
cc1:note: someone does not honour COPTS correctly, passed 18 times
Building C object util/log/Log.c complete
cc1:note: someone does not honour COPTS correctly, passed 18 times
Building C object crypto/random/seed/RandomSeed.c complete
cc1:note: someone does not honour COPTS correctly, passed 18 times
Building C object exception/Except.c complete
cc1:note: someone does not honour COPTS correctly, passed 18 times
Building C object exception/Er.c complete
[... 389 lines cut ...]
/workdir/openwrt/build_dir/target-aarch64_generic_musl/cjdns-cjdns-v21/node_build/builder.js:741
Total build time: 22773ms.
            if (err) { throw err; }
                       ^
Error: gcc -Wl,-z,relro,-z,now,-z,noexecstack -pie -L/workdir/openwrt/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/usr/lib -L/workdir/openwrt/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/lib -znow -zrelro -flto -Os -o build_linux/contrib_c_mkpasswd_c build_linux/util_Assert_c.o,build_linux/util_Bits_c.o,build_linux/memory_Allocator_c.o,build_linux/util_CString_c.o,build_linux/exception_Er_c.o,build_linux/exception_Except_c.o,build_linux/util_log_Log_c.o,build_linux/crypto_random_seed_RandomSeed_c.o,build_linux/crypto_random_seed_DevUrandomRandomSeed_c.o,build_linux/util_Hex_c.o,build_linux/crypto_random_seed_ProcSysKernelRandomUuidRandomSeed_c.o,build_linux/crypto_random_seed_GetEntropyRandomSeed_c.o,build_linux/crypto_random_seed_SystemRandomSeed_c.o,build_linux/crypto_random_Random_c.o,build_linux/memory_MallocAllocator_c.o,build_linux/contrib_c_mkpasswd_c.o build_linux/dependencies/cnacl/jsbuild/libnacl.a,build_linux/dependencies/libuv/out/Release/libuv.a,-lpthread,-lrt
/workdir/openwrt/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/lib/gcc/aarch64-openwrt-linux-musl/11.3.0/../../../../aarch64-openwrt-linux-musl/bin/ld: cannot find build_linux/dependencies/libuv/out/Release/libuv.a: No such file or directory
collect2: error: ld returned 1 exit status
    at error (/workdir/openwrt/build_dir/target-aarch64_generic_musl/cjdns-cjdns-v21/node_build/builder.js:53:15)
    at /workdir/openwrt/build_dir/target-aarch64_generic_musl/cjdns-cjdns-v21/node_build/builder.js:125:22
    at /workdir/openwrt/build_dir/target-aarch64_generic_musl/cjdns-cjdns-v21/node_build/builder.js:95:13
    at ChildProcess.<anonymous> (/workdir/openwrt/build_dir/target-aarch64_generic_musl/cjdns-cjdns-v21/tools/lib/Semaphore.js:7:30)
    at ChildProcess.emit (events.js:198:13)
    at maybeClose (internal/child_process.js:982:16)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:259:5)
make[3]: *** [Makefile:149: /workdir/openwrt/build_dir/target-aarch64_generic_musl/cjdns-cjdns-v21/.built] Error 1
make[3]: Leaving directory '/workdir/openwrt/feeds/routing/cjdns'
time: package/feeds/routing/cjdns/compile#33.89#7.92#23.42
    ERROR: package/feeds/routing/cjdns failed to build.
make[2]: *** [package/Makefile:116: package/feeds/routing/cjdns/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:230: world] Error 2
```
</details>

wfleurant, 2022-06-14:
re: cannot find build_linux/dependencies/libuv/out/Release/libuv.a: No such file or directory

zhangguanzhang, 2022-06-15:
> i need more information please. are you building development branch or release branch? it works in develop branch -- please share more information.
> 
> ```
> time: target/linux/install#26.38#16.20#7.46
> Generating package index...
> Generating index for package ./cjdns_v21-3_aarch64_cortex-a72.ipk
> Signing package index...
> ```

I used the master branch to build

wfleurant, 2022-06-15:
OK i built the same target, it could be something else going on. Do you want to share the top part of your .config file?
I am trying to reproduce the problem. I may need the exact make/model build you are targeting...

Sooner or later we should be introducing cjdns based on Rust -- are you actively testing / targeting? I am looking for active users/testers.

zhangguanzhang, 2022-06-15:
[config.zip](https://github.com/openwrt/routing/files/8910875/config.zip)
here is my `.config`

wfleurant, 2022-06-16:
Thanks, cool dot config.  

I was able to compile successfully using Debian GNU/Linux 11 (bullseye).  
Compiled https://git.openwrt.org/openwrt/openwrt.git  

I may have overlooked something. Is it something to do with your Python environment?  
Your system has Python3? I compiled with `3.9.2`.

```
Build Libuv
Traceback (most recent call last):
  File "/workdir/openwrt/build_dir/target-aarch64_generic_musl/cjdns-cjdns-v21/build_linux/dependencies/libuv/./gyp_uv.py", line 14, in <module>
    import gyp
  File "/workdir/openwrt/build_dir/target-aarch64_generic_musl/cjdns-cjdns-v21/build_linux/dependencies/libuv/build/gyp/pylib/gyp/__init__.py", line 10, in <module>
    import gyp.input
  File "/workdir/openwrt/build_dir/target-aarch64_generic_musl/cjdns-cjdns-v21/build_linux/dependencies/libuv/build/gyp/pylib/gyp/input.py", line 8, in <module>
    import gyp.common
  File "/workdir/openwrt/build_dir/target-aarch64_generic_musl/cjdns-cjdns-v21/build_linux/dependencies/libuv/build/gyp/pylib/gyp/common.py", line 497, in <module>
    class OrderedSet(collections.MutableSet):
AttributeError: module 'collections' has no attribute 'MutableSet'
make[4]: *** out: No such file or directory.  Stop.
```

zhangguanzhang, 2022-06-17:
> Thanks, cool dot config.
>
> I was able to compile successfully using Debian GNU/Linux 11 (bullseye). Compiled https://git.openwrt.org/openwrt/openwrt.git
>
> I may have overlooked something. Is it [...]

I build on ubuntu-20.04, and installed them:
```
build-essential asciidoc binutils bzip2 gawk gettext git libncurses5-dev libz-dev \
  patch python3 python2.7 unzip zlib1g-dev  libc6-dev-i386 subversion \
  flex uglifyjs git-core gcc-multilib p7zip p7zip-full msmtp libssl-dev texinfo libglib2.0-dev xmlto \
  qemu-utils upx libelf-dev autoconf automake libtool autopoint \
  device-tree-compiler g++-multilib antlr3 gperf wget curl swig rsync
```

wfleurant, 2022-06-17:
I compiled on the same setup we describe here.

I see a difference... It appears your build is using Python 3.10 while mine is using Python 3.8

This is a new bug. Thank you. I will look to see what can be patched.  

my build
```
done
testing python python3.7
error starting python Error: spawn python3.7 ENOENT
testing python python3
sys.version_info(major=3, minor=8, micro=10, releaselevel='final', serial=0)

Build Libuv
['-Dtarget_arch=aarch64', '-f', 'make-linux', '/dev/shm/openwrt/build_dir/target-aarch64_generic_musl/cjdns-cjdns-v21/build_linux/dependencies/libuv/uv.gyp', '-I', '/dev/shm>
/dev/shm/openwrt/build_dir/target-aarch64_generic_musl/cjdns-cjdns-v21/build_linux/dependencies/libuv/./build/gyp/pylib/gyp/input.py:1184: SyntaxWarning: "is" with a literal>
  if the_dict_key is 'variables' and variable_name in the_dict:
make[3]: Entering directory '/dev/shm/openwrt/build_dir/target-aarch64_generic_musl/cjdns-cjdns-v21/build_linux/dependencies/libuv/out'
  ccache_cc '-D_LARGEFILE_SOURCE' '-D_FILE_OFFSET_BITS=64' '-DHAVE_CONFIG_H' -I../include -I../src  -pthread -Wall -ansi -pthread -fvisibility=hidden -g --std=gnu89 -pedanti>
  ccache_cc '-D_LARGEFILE_SOURCE' '-D_FILE_OFFSET_BITS=64' '-DHAVE_CONFIG_H' -I../include -I../src  -pthread -Wall -ansi -pthread -fvisibility=hidden -g --std=gnu89 -pedanti>
...
time: package/feeds/routing/cjdns/compile#27.98#5.32#5.92
make[1]: Leaving directory '/dev/shm/openwrt'
```

your build
```
testing python python3.7
error starting python Error: spawn python3.7 ENOENT
testing python python3
sys.version_info(major=3, minor=10, micro=4, releaselevel='final', serial=0)
Build Libuv
Traceback (most recent call last):
  File "/workdir/openwrt/build_dir/target-aarch64_generic_musl/cjdns-cjdns-v21/build_linux/dependencies/libuv/./gyp_uv.py", line 14, in <module>
    import gyp
  File "/workdir/openwrt/build_dir/target-aarch64_generic_musl/cjdns-cjdns-v21/build_linux/dependencies/libuv/build/gyp/pylib/gyp/__init__.py", line 10, in <module>
    import gyp.input
  File "/workdir/openwrt/build_dir/target-aarch64_generic_musl/cjdns-cjdns-v21/build_linux/dependencies/libuv/build/gyp/pylib/gyp/input.py", line 8, in <module>
    import gyp.common
  File "/workdir/openwrt/build_dir/target-aarch64_generic_musl/cjdns-cjdns-v21/build_linux/dependencies/libuv/build/gyp/pylib/gyp/common.py", line 497, in <module>
    class OrderedSet(collections.MutableSet):
AttributeError: module 'collections' has no attribute 'MutableSet'
make[4]: *** out: No such file or directory.  Stop.
```

zhangguanzhang, 2022-06-17:
woow, nice catch. 👍 
But I can't figure out why it compiles without python in the staging_dir directory

wfleurant, 2022-06-17:
okay.... we can probably patch frozen package dependency `gyp`  
basically, replace `node_build/dependencies/libuv/build/gyp/` with latest (https://chromium.googlesource.com/external/gyp)  
and commending out Werror in `node_build/make.js`

im out the door, i'll try to patch this weekend.. thanks again for reporting

```
% ./cjdroute  --version
Cjdns version: cjdns-v21.2-7-g7bc76859-dirty
Cjdns protocol version: 21

% python3 -V
Python 3.10.4
```

wfleurant, 2022-06-17:
for now, lets see if this works.  
decompress this patch, put it here like `package/feeds/routing/cjdns/patches/020-latest-gyp-no-werror.patch` 
then try `make V=99 package/cjdns/{clean,compile}`

[020-latest-gyp-no-werror.patch.gz](https://github.com/openwrt/routing/files/8931018/020-latest-gyp-no-werror.patch.gz)

```
time: package/kernel/linux/compile#1.11#0.05#1.16
make[2]: Entering directory '/dev/shm/openwrt/feeds/routing/cjdns'
touch /dev/shm/openwrt/build_dir/target-aarch64_generic_musl/cjdns-cjdns-v21/.prepared_c02167bb5e56b364300f16e702ea4aa0_6664517399ebbbc92a37c5bb081b5c53_check
. /dev/shm/openwrt/include/shell.sh; gzip -dc /dev/shm/openwrt/dl/cjdns-v21.tar.gz | tar -C /dev/shm/openwrt/build_dir/target-aarch64_generic_musl/cjdns-cjdns-v21/.. -xf -
[ ! -d ./src/ ] || cp -fpR ./src/. /dev/shm/openwrt/build_dir/target-aarch64_generic_musl/cjdns-cjdns-v21

Applying ./patches/001-five-mins-builder-zonk.patch using plaintext: 
patching file node_build/builder.js

Applying ./patches/010-musl-12x.patch using plaintext: 
patching file memory/Allocator.c

Applying ./patches/020-latest-gyp-no-werror.patch using plaintext: 
patching file node_build/dependencies/libuv/build/gyp/gyp
patching file node_build/dependencies/libuv/build/gyp/gyp_main.py
....
Build completed successfully, type ./cjdroute to begin setup.
Total build time: 5277ms.
Packaged contents of /dev/shm/openwrt/build_dir/target-aarch64_generic_musl/cjdns-cjdns-v21/ipkg-aarch64_generic/cjdns into /dev/shm/openwrt/bin/packages/aarch64_generic/routing/cjdns_v21-3_aarch64_generic.ipk
```

zhangguanzhang, 2022-06-18:
I will try it tomorrow

zhangguanzhang, 2022-06-19:
used the commit https://github.com/openwrt/routing/commit/217330bb5e7c17cc2e9f116f8a853f2e5fb86697.patchhttps://github.com/openwrt/routing/commit/217330bb5e7c17cc2e9f116f8a853f2e5fb86697.patch
will be ok.
Could you cherry-pick it to `21.02`？

wfleurant, 2022-06-19:
trying to test / verify luci-app-cjdns and cjdroute configuration is setting/getting from UCI correctly.
I think v21.1 has additional features like supernode and.... Lets just test a bit before a backport.

wfleurant, 2022-06-19:
let me know if the compiling is fixed, we can close and then backport v21.1 + supernodes to 21.02, 22.03.  
thanks for contributing your help and testing.

zhangguanzhang, 2022-06-19:
> let me know if the compiling is fixed, we can close and then backport v21.1 + supernodes to 21.02, 22.03. thanks for contributing your help and testing.

I tested it on the branch 21.02, It works fine