openwrt/routing #899: `oonf-olsrd2`: Build fails for me with `FAILED: src/olsrd2/CMakeFiles/olsrd2_static /[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/oonf-olsrd2-v0.15.1/src/olsrd2/CMakeFiles/olsrd2_static` on 22.03.

felixbecker2, 2022-07-15:
### Maintainer:

Henning Rogge <hrogge(at)gmail.com>, @HRogge

### Environment:

* Building for Xunlog OrangePi Plus (sun8i):  
  `grep ^CONFIG_TARGET .config`:
  ```
  CONFIG_TARGET_sunxi=y
  CONFIG_TARGET_sunxi_cortexa7=y
  CONFIG_TARGET_sunxi_cortexa7_DEVICE_xunlong_orangepi-plus=y
  CONFIG_TARGET_BOARD="sunxi"
  CONFIG_TARGET_SUBTARGET="cortexa7"
  CONFIG_TARGET_PROFILE="DEVICE_xunlong_orangepi-plus"
  CONFIG_TARGET_ARCH_PACKAGES="arm_cortex-a7_neon-vfpv4"
  [...]
  ```
* OpenWrt version:
  - Git branch `openwrt-22.03` from `git://git.openwrt.org/openwrt/openwrt.git`,
  - initially checked out on 2022-07-05,
  - latest `git pull` and `./scripts/feeds update -a && ./scripts/feeds update -a` on 2022-07-14, ca. 12:00 UTC  
    *(I give up on posting latest commit date and hash, since `git log` gives me some old dates, and I don't know how I can make `git log` printing out the last operation. I don't know much about `git`.)*
* Building on GNU/Linux on x86_64-architecture (Distribution: [Artix Linux](http://artixlinux.org)).
* OpenWrt configuration: [&rarr;&nbsp;`config.txt`](https://github.com/openwrt/routing/files/9120508/config.oonf-olsrd2-build-failure.txt).
* `grep -i oonf .config`:  
  ```
  CONFIG_PACKAGE_oonf-dlep-proxy=m
  CONFIG_PACKAGE_oonf-dlep-radio=m
  CONFIG_PACKAGE_oonf-init-scripts=m
  CONFIG_PACKAGE_oonf-olsrd2=m
  CONFIG_OONF_NHDP_AUTOLL4=y
  CONFIG_OONF_OLSRV2_LAN_IMPORT=y
  CONFIG_OONF_OLSRV2_ROUTE_MODIFIER=y
  CONFIG_OONF_GENERIC_DLEP_ROUTER=y
  CONFIG_OONF_GENERIC_REMOTECONTROL=y
  CONFIG_OONF_GENERIC_HTTP=y
  CONFIG_OONF_OLSRV2_MPR=y
  ```
* `grep -i olsr .config | grep -v luci-i18n`:  
  ```
  CONFIG_PACKAGE_luci-app-olsr=y
  CONFIG_PACKAGE_luci-app-olsr-services=y
  CONFIG_PACKAGE_luci-app-olsr-viz=y
  # OLSR.org network framework
  CONFIG_PACKAGE_oonf-olsrd2=m
  CONFIG_OONF_OLSRV2_LAN_IMPORT=y
  CONFIG_OONF_OLSRV2_ROUTE_MODIFIER=y
  CONFIG_OONF_OLSRV2_MPR=y
  # end of OLSR.org network framework
  CONFIG_PACKAGE_olsrd=y
  CONFIG_PACKAGE_olsrd-mod-arprefresh=y
  CONFIG_PACKAGE_olsrd-mod-bmf=y
  CONFIG_PACKAGE_olsrd-mod-dot-draw=y
  CONFIG_PACKAGE_olsrd-mod-dyn-gw=y
  CONFIG_PACKAGE_olsrd-mod-dyn-gw-plain=y
  CONFIG_PACKAGE_olsrd-mod-filtergw=y
  CONFIG_PACKAGE_olsrd-mod-httpinfo=y
  CONFIG_PACKAGE_olsrd-mod-jsoninfo=y
  CONFIG_PACKAGE_olsrd-mod-mdns=y
  CONFIG_PACKAGE_olsrd-mod-nameservice=y
  CONFIG_PACKAGE_olsrd-mod-netjson=y
  CONFIG_PACKAGE_olsrd-mod-p2pd=y
  CONFIG_PACKAGE_olsrd-mod-pgraph=y
  CONFIG_PACKAGE_olsrd-mod-pud=y
  CONFIG_PACKAGE_olsrd-mod-quagga=y
  CONFIG_PACKAGE_olsrd-mod-secure=y
  CONFIG_PACKAGE_olsrd-mod-sgwdynspeed=y
  CONFIG_PACKAGE_olsrd-mod-txtinfo=y
  CONFIG_PACKAGE_olsrd-mod-watchdog=y
[... 61 lines cut ...]
time: package/libs/libubox/compile#0.42#0.11#0.47
make[2]: Entering directory '/[...]/openwrt-22.03/package/system/uci'
echo "libuci" >> /[...]/openwrt-22.03/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/pkginfo/uci.default.install
echo "libuci-lua" >> /[...]/openwrt-22.03/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/pkginfo/uci.default.install
echo "uci" >> /[...]/openwrt-22.03/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/pkginfo/uci.default.install
make[2]: Leaving directory '/[...]/openwrt-22.03/package/system/uci'
time: package/system/uci/compile#0.37#0.11#0.44
make[2]: Entering directory '/[...]/openwrt-22.03/feeds/routing/oonf-olsrd2'
rm -f /[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/oonf-olsrd2-v0.15.1/.built
touch /[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/oonf-olsrd2-v0.15.1/.built_check
MAKEFLAGS="" /[...]/openwrt-22.03/staging_dir/host/bin/ninja -v -j1 -C /[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/oonf-olsrd2-v0.15.1 olsrd2_static
ninja: Entering directory `/[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/oonf-olsrd2-v0.15.1'
[1/1] cd /[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/oonf-olsrd2-v0.15.1/src/olsrd2 && false
FAILED: src/olsrd2/CMakeFiles/olsrd2_static /[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/oonf-olsrd2-v0.15.1/src/olsrd2/CMakeFiles/olsrd2_static 
cd /[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/oonf-olsrd2-v0.15.1/src/olsrd2 && false
ninja: build stopped: subcommand failed.
make[2]: *** [Makefile:91: /[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/oonf-olsrd2-v0.15.1/.built] Error 1
make[2]: Leaving directory '/[...]/openwrt-22.03/feeds/routing/oonf-olsrd2'
time: package/feeds/routing/oonf-olsrd2/compile#0.35#0.06#0.37
    ERROR: package/feeds/routing/oonf-olsrd2 failed to build.
make[1]: *** [package/Makefile:116: package/feeds/routing/oonf-olsrd2/compile] Error 1
make[1]: Leaving directory '/[...]/openwrt-22.03'
make: *** [/[...]/openwrt-22.03/include/toplevel.mk:230: package/feeds/routing/oonf-olsrd2/compile] Error 2
```
</details>

mwarning, 2023-02-23:
@PolynomialDivision any idea?

XDjackieXD, 2023-02-24:
To fix builds for me I updated olsrd2 to latest develop branch, as the master branch is pretty much unused currently (and olsrd2 does not really have any stable release. latest develop is the most stable you'll get as there were quite some fixes merged).

Attached is a patch that applies to the routing feed version that is currently used by the latest 22.03 stable.
[olsrd2.zip](https://github.com/openwrt/routing/files/10827607/olsrd2.zip) (zip because for whatever reason github won't let me upload a patch file with an unsupported filetype error...)

PolynomialDivision, 2023-02-24:
Why don't you add a PR?

XDjackieXD, 2023-02-24:
Haven't had time so far. I was planning on making a PR next week

XDjackieXD, 2023-03-04:
https://github.com/openwrt/routing/pull/956 PR opened