openwrt/routing #908: cjdns: Segmentation Fault on OpenWRT 22.03

ringtailedfox, 2022-10-08:
Maintainer: @\[wfleurant] (https://github.com/wfleurant) (find it by checking history of the package Makefile)
Environment: (put here ARMv7 / ipq806x, Netgear R7800, OpenWrt 22.03)

Description:
cjdns cannot be configured because the program segfaults.

```
root@Nighthawk:/etc# cjdroute --genconf > /etc/cjdroute.conf
```

In all honesty, running cjdrouteconf with ANY switches, including none at all, resulgs in SIGSEV (Segmentation Fault).

```
root@Nighthawk:/usr/bin# cat /etc/cjdroute.conf | cjdrouteconf set
lua: /usr/lib/lua/cjdns/uci.lua:141: attempt to index field 'admin' (a nil value)
stack traceback:
        /usr/lib/lua/cjdns/uci.lua:141: in function 'set'
        /usr/bin/cjdrouteconf:23: in main chunk
        [C]: ?
root@Nighthawk:/usr/bin# root@Nighthawk:/usr/bin# cat /etc/cjdroute.conf | cjdro
uteconf set
stack traceback:
        /usr/lib/lua/cjdns/uci.lua:141: in function 'set'
        /usr/bin/cjdrouteconf:23: in main chunk
        [C]: ?
```

```
root@Nighthawk:/usr/bin# cat /etc/cjdroute.conf | cjdrouteconf set
lua: /usr/lib/lua/cjdns/uci.lua:141: attempt to index field 'admin' (a nil value)
stack traceback:
        /usr/lib/lua/cjdns/uci.lua:141: in function 'set'
        /usr/bin/cjdrouteconf:23: in main chunk
        [C]: ?
```

running GDB gives a similar result:

```
root@Nighthawk:/etc# gdb $(which cjdroute) -ex 'r --genconf'
GNU gdb (GDB) 11.2
Copyright (C) 2022 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "arm-openwrt-linux".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /usr/sbin/cjdroute...
(No debugging symbols found in /usr/sbin/cjdroute)
Starting program: /usr/sbin/cjdroute --genconf
warning: Unable to find dynamic linker breakpoint function.
GDB will be unable to debug shared library initializers
and track explicitly loaded dynamic code.

Program received signal SIGSEGV, Segmentation fault.
0xb6fd90bc in ?? ()
(gdb) Program received signal SIGSEGV, Segmentation fault.
Quit
(gdb) bt
#0  0xb6fd90bc in ?? ()
#1  0xb6fd9840 in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb) Reading symbols from /usr/sbin/cjdroute...
(No debugging symbols found in /usr/sbin/cjdroute)
Starting program: /usr/sbin/cjdroute --genconf
warning: Unable to find dynamic linker breakpoint function.
GDB will be unable to debug shared library initializers
and track explicitly loaded dynamic code.

Program received signal SIGSEGV, Segmentation fault.
Undefined command: "Reading".  Try "help".
(gdb) quit
A debugging session is active.

        Inferior 1 [process 28290] will be killed.

Quit anyway? (y or n) y
```

This might be why the service/daemon for CJDNS crashes when it tries to start:
```
root@Nighthawk:/usr/bin# /etc/init.d/cjdns start
Segmentation fault
Segmentation fault
dkjson: no valid JSON value (reached the end) (try cjdroute --cleanconf)
```

When i looked at that lua file (/usr/lib/lua/cjdns/uci.lua), this is what line 141 says:

```
local admin_address, admin_port = string.match(obj.admin.bind, "^(.*):(.*)$")
```

As for what line 23 in /usr/bin/cjdrouteconf ?  That's below:

```
    cjdns.uci.set(obj)
```

None of this makes any sense to me... can anyone else understand this?

PolynomialDivision, 2022-10-09:
@wfleurant

wfleurant, 2022-10-12:
yes.. seems duplicate from last time toolchain patches. give me a minute to catch up

wfleurant, 2022-10-30:
i am really bogged down to figure this out at the moment. anyone have a hint? is it another compiler update like before?

predators46, 2022-11-05:
what changed in openwrt 22.03.x? i had the same problem, i created a file in openwrt 22.03.x and got "Segmentation Fault" error if i created it on openwrt 21.02.x everything went smoothly

ringtailedfox, 2024-02-29:
Okay, I do have an update to this!

I managed to re-compile cjdns from source for OpenWrt 23.05.2, r23630-842932a63d .... it no longer segfaults during install, but it still segfaults when i attempt to run it. sadly, gdb says there's no debug symbols for it this time.

output when i run gdb:
root@Nighthawk:~# gdb $(which cjdroute) -ex 'r --genconf'
GNU gdb (GDB) 12.1
Copyright (C) 2022 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "arm-openwrt-linux".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /usr/sbin/cjdroute...
(No debugging symbols found in /usr/sbin/cjdroute)
Starting program: /usr/sbin/cjdroute --genconf
warning: Unable to find dynamic linker breakpoint function.
GDB will be unable to debug shared library initializers
and track explicitly loaded dynamic code.

Program received signal SIGSEGV, Segmentation fault.
0xb6fe84f8 in ?? ()

(gdb) Quit

root@Nighthawk:~# file /usr/sbin/cjdroute
/usr/sbin/cjdroute: ELF 32-bit LSB pie executable, ARM, EABI5 version 1 (SYSV), dynamically linked, interpreter /lib/ld-musl-armhf.so.1, no section header

ringtailedfox, 2025-09-06:
[Got a big update for CJDNS on OpenWRT](https://github.com/openwrt/routing/issues/1048#issuecomment-3262673427)