openwrt/routing #920: bird1: out of memory when executing /etc/init.d/bird4

rogerpueyo, 2022-11-16:
Hi,

It looks like the initd script at `/etc/init.d/bird4` shipped with package `bird1-ipv4-uci` (which is actually a symlink to `/etc/bird4/init.d/bird4`) makes the system run out of memory.

**Setup**:
Fresh OpenWrt 22.03.2 running on an x86_64 VM with plenty of RAM (2 GB).

**Steps to reproduce**:

1. Update package feeds (`opkg update`)
2. Install package `bird1-ipv4` (`opkg install bird1-ipv4` => `Package bird1-ipv4 (1.6.8-2) installed in root is up to date.`)
3. Start `bird4` without changing any setting (`/etc/init.d/bird4 start` => Nothing happens)
4. Install package `bird1-ipv4-uci` (`opkg install bird1-ipv4-uci` => `Package bird1-ipv4-uci (0.3-1) installed in root is up to date.)
5. Start `bird4` without changing any setting (`/etc/init.d/bird4 start` => `Starting bird4 Service [ ... ]` => Out of memory and tons of `/etc/rc.common: export: line 136: : bad variable name` messages)

Some more details:

`ps` shows hundreds of instances of the initd script:
```
 5552 root      1392 S    {bird4} /bin/sh /etc/rc.common /etc/init.d/bird4 start
 5566 root      1392 S    {bird4} /bin/sh /etc/rc.common /etc/init.d/bird4 start
 5567 root      1392 S    {bird4} /bin/sh /etc/rc.common /etc/init.d/bird4 start
 5568 root      1392 S    {bird4} /bin/sh /etc/rc.common /etc/init.d/bird4 start
 5569 root      1392 S    {bird4} /bin/sh /etc/rc.common /etc/init.d/bird4 start
 ... (> 400 more lines)
 5955 root      2676 S    {bird4} /bin/sh /etc/rc.common /etc/init.d/bird4 start
 5957 root      2680 S    {bird4} /bin/sh /etc/rc.common /etc/init.d/bird4 start
 5958 root      2684 S    {bird4} /bin/sh /etc/rc.common /etc/init.d/bird4 start
 5959 root      1128 R    ps
 5960 root      2684 S    {bird4} /bin/sh /etc/rc.common /etc/init.d/bird4 start
 5961 root      2688 S    {bird4} /bin/sh /etc/rc.common /etc/init.d/bird4 start
 5962 root      2692 S    {bird4} /bin/sh /etc/rc.common /etc/init.d/bird4 start
 5963 root      2696 R    {bird4} /bin/sh /etc/rc.common /etc/init.d/bird4 start
```

`dmesg` shows the system ran out of memory:
```
[  584.370228] bird4 invoked oom-killer: gfp_mask=0xcc0(GFP_KERNEL), order=0, oom_score_adj=0
[  584.370876] CPU: 0 PID: 6542 Comm: bird4 Not tainted 5.10.146 #0
[  584.371255] Hardware name: innotek GmbH VirtualBox/VirtualBox, BIOS VirtualBox 12/01/2006
[  584.371789] Call Trace:
[  584.372019]  0xffffffff819f49c6
[  584.372297]  0xffffffff819ed7d6
[  584.372609]  0xffffffff819ed9ae
[  584.372843]  0xffffffff811abd9c
[  584.373076]  0xffffffff811f07bf
[  584.373310]  ? 0xffffffff811feb76
[  584.373551]  0xffffffff811f0ae5
[  584.373797]  0xffffffff811fd033
[  584.374040]  0xffffffff811fdf93
[  584.374273]  0xffffffff811febdd
[  584.374507]  0xffffffff811e55f9
[  584.374741]  0xffffffff811e569d
[  584.374973]  0xffffffff810b7f46
[  584.375250]  0xffffffff810b9872
[  584.375484]  0xffffffff810b9d89
[  584.375717]  ? 0xffffffff81230771
[  584.375958]  0xffffffff810ba226
[  584.376191]  0xffffffff81a2d3e8
[  584.376424]  0xffffffff81c000a9
[  584.376657] RIP: 0033:0x7f5f9026c7bb
[  584.376916] Code: Unable to access opcode bytes at RIP 0x7f5f9026c791.
[  584.377807] RSP: 002b:00007ffec5c4de30 EFLAGS: 00000246 ORIG_RAX: 0000000000000039
[  584.378267] RAX: ffffffffffffffda RBX: 0000000000000084 RCX: 00007f5f9026c7bb
[  584.378706] RDX: 00007f5f9027455c RSI: 0000000000000000 RDI: 00007f5f902a3fc4
[  584.379152] RBP: 00007ffec5c4de30 R08: 00007f5f901732c0 R09: 0000000000000002
[  584.379555] R10: 0000000000000008 R11: 0000000000000246 R12: 00007f5f902a3fc4
[  584.379957] R13: 00007ffec5c4dee0 R14: 00007f5f902a3b48 R15: 00007f5f90213b00
[  584.380361] Mem-Info:
[  584.380545] active_anon:258 inactive_anon:21820 isolated_anon:0
[  584.380545]  active_file:6 inactive_file:6 isolated_file:0
[  584.380545]  unevictable:0 dirty:0 writeback:0
[  584.380545]  slab_reclaimable:621 slab_unreclaimable:204065
[  584.380545]  mapped:2 shmem:245 pagetables:8288 bounce:0
[  584.380545]  free:5014 free_pcp:347 free_cma:0
[  584.382183] Node 0 active_anon:1032kB inactive_anon:87280kB active_file:24kB inactive_file:24kB unevictable:0kB isolated(anon):0kB isolated(file):0kB mapped:8kB dirty:0kB writeback:0kB shmem:980kB writeback_tmp:0kB kernel_stack:16240kB all_unreclaimable? yes
[  584.383467] DMA free:4112kB min:256kB low:320kB high:384kB reserved_highatomic:0KB active_anon:0kB inactive_anon:548kB active_file:0kB inactive_file:0kB unevictable:0kB writepending:0kB present:15992kB managed:15908kB mlocked:0kB pagetables:204kB bounce:0kB free_pcp:0kB local_pcp:0kB free_cma:0kB
[  584.384838] lowmem_reserve[]: 0 966 966 966
[  584.385108] DMA32 free:15944kB min:16124kB low:20152kB high:24180kB reserved_highatomic:0KB active_anon:1032kB inactive_anon:86732kB active_file:24kB inactive_file:24kB unevictable:0kB writepending:0kB present:1032128kB managed:994940kB mlocked:0kB pagetables:32948kB bounce:0kB free_pcp:1388kB local_pcp:1388kB free_cma:0kB
[  584.386894] lowmem_reserve[]: 0 0 0 0
[  584.387182] DMA: 0*4kB 2*8kB (UM) 2*16kB (UM) 1*32kB (U) 1*64kB (M) 1*128kB (M) 1*256kB (M) 1*512kB (U) 1*1024kB (M) 1*2048kB (M) 0*4096kB = 4112kB
[  584.387965] DMA32: 92*4kB (ME) 95*8kB (UME) 26*16kB (ME) 12*32kB (UME) 11*64kB (E) 4*128kB (UME) 0*256kB 1*512kB (U) 0*1024kB 0*2048kB 3*4096kB (UM) = 15944kB
[  584.389355] 257 total pagecache pages
[  584.389764] 0 pages in swap cache
[  584.390076] Swap cache stats: add 0, delete 0, find 0/0
[  584.390430] Free swap  = 0kB
[  584.390651] Total swap = 0kB
[  584.390873] 262030 pages RAM
[  584.391110] 0 pages HighMem/MovableOnly
[  584.391376] 9318 pages reserved
[  584.391610] Tasks state (memory values in pages):
[  584.391916] [  pid  ]   uid  tgid total_vm      rss pgtables_bytes swapents oom_score_adj name
[  584.392458] [   1122]    81  1122      241       30    45056        0             0 ubusd
[  584.392967] [   1124]     0  1124      282       16    45056        0             0 ash
[  584.393467] [   1155]     0  1155      203       16    36864        0             0 urngd
[  584.393984] [   1417]     0  1417      871       45    49152        0             0 uhttpd
[  584.394485] [   1649]   514  1649      248       21    36864        0             0 logd
[  584.394979] [   1703]     0  1703      335       42    40960        0             0 rpcd
[  584.395525] [   1914]     0  1914      234       16    45056        0             0 dropbear
[  584.396041] [   2030]     0  2030      329       42    40960        0             0 netifd
[  584.396528] [   2107]     0  2107      300       29    40960        0             0 odhcpd
[  584.397015] [   2149]     0  2149      208        9    40960        0             0 odhcp6c
[  584.397514] [   2153]     0  2153      280        8    40960        0             0 udhcpc
[  584.398003] [   2679]     0  2679      540       28    40960        0             0 ntpd
[  584.398484] [   2691]   123  2691      280        9    36864        0             0 ntpd
[  584.398964] [   2902]     0  2902      540       28    45056        0             0 dnsmasq
[  584.399508] [   2905]   453  2905      295       31    45056        0             0 dnsmasq
[  584.399999] [   4076]     0  4076      239       18    45056        0             0 dropbear
[  584.400493] [   4081]     0  4081      282       17    36864        0             0 ash
[  584.400969] [   5552]     0  5552      348       76    45056        0             0 bird4
[  584.401489] [   5566]     0  5566      348       77    45056        0             0 bird4
[  584.402040] [   5567]     0  5567      348       77    45056        0             0 bird4
...
[  584.889579] [   6538]     0  6538     1170      917    49152        0             0 bird4
[  584.890077] [   6539]     0  6539     1171      918    49152        0             0 bird4
[  584.890561] [   6540]     0  6540     1172      919    49152        0             0 bird4
[  584.891090] [   6541]     0  6541     1173      920    49152        0             0 bird4
[  584.891574] [   6542]     0  6542     1173      922    49152        0             0 bird4
[  584.892057] oom-kill:constraint=CONSTRAINT_NONE,nodemask=(null),cpuset=/,mems_allowed=0,global_oom,task_memcg=/,task=bird4,pid=6542,uid=0
[  584.892738] Out of memory: Killed process 6542 (bird4) total-vm:4692kB, anon-rss:3688kB, file-rss:0kB, shmem-rss:0kB, UID:0 pgtables:48kB oom_score_adj:0
```

If I replace the `/etc/init.d/bird4` symlink shipped with package `bird1-ipv4-uci` with the original file from `bird1-ipv4`, there seems to be no problem and `bird4` is actually started/stopped.

I guess @tohojo is the package maintainer. Sorry to bother you.

tohojo, 2022-11-16:
Hmm, I guess the Bird uci config layer bitrotted to the point where it no longer works? I'm not actually the maintainer of that bit, looks like that's @eloicaso :)

In the meantime, I'd suggest just not using the bird1-ipv4-uci package; the base package should work just fine, you just need to write your own config file.

Or even better, move to bird2; the bird1 branch is unmaintained upstream, so I'd really recommend moving off of it...

rogerpueyo, 2022-11-16:
Hi, @tohojo ,

Thanks for your reply. You're right. I could trace the issue down to the `config_get_bool` UCI call in `/etc/bird4/init.d/bird4-lib.sh` that no longer works.

I wonder if @eloicaso would like to take a look at it (Hi, Eloi! Long time no see, I hope you're doing well :-))

Otherwise, I'll try to figure out a quick workaround and consider moving towards bird2.

rogerpueyo, 2022-11-16:
Hi,

I think I found the culprit: the `/etc/bird4/init.d/bird4-lib.sh` file implements a `get_bool()` function which, I believe, collides with a function named the same in `/lib/functions.sh`.

Maybe this function didn't exist back when the package was created and @eloicaso implemented it.

I'll be submitting a patch, @tohojo.

Cheers!

commit d32f93e0 (2023-03-06): bird1: rename get_bool() to get_a_bool() in bird{4,6}-lib.sh

This patch renames the get_bool() function, implemented in the
bird4-lib.sh and bird6-lib.sh files (respectively, for the IPv4 and
the IPv6 versions of the package), as well as all the calls to it.
This way, we avoid a function name collision with the one provided by
file /lib/functions.sh, which is slightly different and caused an
out-of-memory error when parsing the bird4/bird6 UCI config files.

Fixes #920.
PR #959 (2023-03-09): Backport bird1-openwrt fixes to 21.02
Maintainer: @rogerpueyo
Compile and run-tested: x86_64 VirtualBox VM, [OpenWrt 21.02 as of 2023/03/09](https://github.com/openwrt/openwrt/commit/23c86d44bcfa00255c32196458b0a47dfa6732d5)

Description: four commits were recently added to the `master` branch, to fix three issues in the `bird1-openwrt` package:
1. 914c27aeb66b21c7510e376625e9ec386a56ee35 and 6697ca291f97438e937897d73acc46a29d7dc9f4 (fix CI tests errors)
2. d32f93e0499164ac171228bbb6e8a5b17b8785fe (fix #920)
3. c37fbab8f5e3af0843cf1bef5aa199a0c0db245b (fix #922)

These three issues also affect [... cut]
commit 258826e8 (2023-03-09): bird1: rename get_bool() to get_a_bool() in bird{4,6}-lib.sh

This patch renames the get_bool() function, implemented in the
bird4-lib.sh and bird6-lib.sh files (respectively, for the IPv4 and
the IPv6 versions of the package), as well as all the calls to it.
This way, we avoid a function name collision with the one provided by
file /lib/functions.sh, which is slightly different and caused an
out-of-memory error when parsing the bird4/bird6 UCI config files.

Fixes #920.

(cherry picked from commit d32f93e0499164ac171228bbb6e8a5b17b8785fe)
commit 3d7df594 (2023-03-09): bird1: rename get_bool() to get_a_bool() in bird{4,6}-lib.sh

This patch renames the get_bool() function, implemented in the
bird4-lib.sh and bird6-lib.sh files (respectively, for the IPv4 and
the IPv6 versions of the package), as well as all the calls to it.
This way, we avoid a function name collision with the one provided by
file /lib/functions.sh, which is slightly different and caused an
out-of-memory error when parsing the bird4/bird6 UCI config files.

Fixes #920.

(cherry picked from commit d32f93e0499164ac171228bbb6e8a5b17b8785fe)
closed 2023-03-06 by BKPepe