openwrt/routing #963: luci-app-bmx6: jquery is outdated and vulnerable

BKPepe, 2023-04-11:
Maintainer: @p4u 
Environment: all

Description:

Hey, a few minutes I got security alert digest from GitHub about outdated ``jquery-1.4.js`` as it is vulnerable to [CVE-2020-11022](https://nvd.nist.gov/vuln/detail/cve-2020-11022). Could you please check it?

BKPepe, 2023-04-25:
Another report came out with this vulnerability.

PolynomialDivision, 2023-04-25:
Maybe we should move this package to [aboned](https://github.com/openwrt/packages-abandoned)?

BKPepe, 2023-09-28:
Do we really want to move these packages, which are vulnerable to abandoned feed to keep continuing users to give a choice to install this package? Well, I don't think so. Anyone who wants to use it can go through the Git history, revert it and update it, but reviving old projects does not make any sense, when there is replacement.

closed 2023-09-29 by BKPepe