openwrt/routing #966: luci-app-babeld: Missing neighbour

marek22k, 2023-04-28:


jow-, 2023-04-28:
@PolynomialDivision

PolynomialDivision, 2023-04-28:
> It's a bug in the JSON output produced by https://github.com/openwrt/routing/blob/master/babeld/src/ubus.c - note the duplicate dictionary key `fe80::2923` - we can't fix this on the LuCI side.

Yes. You are right. We could change it to `[IP]%dev` to fix duplicate addressing?

jow-, 2023-05-11:
> We could change it to [IP]%[DEV] to fix duplicate addressing

Sounds sensible to me!

PolynomialDivision, 2023-05-13:
I think it is possible that this still brings some duplicate entries with it. Maybe @jech can say something to it, how we can do the sorting in the best manner?

marek22k, 2023-05-13:
Actually, a link-local IP address must not occur twice on an interface. The DAD should exist for this purpose. Alternatively, the whole thing could perhaps be represented as an array?

jech, 2023-05-13:
Right.  Neighbours are indexed by (adress, interface).  It is perfectly normal to see the same neighbour on two interfaces: if a router A has two radios tuned at the same frequency, or two Ethernet interfaces connected to a single switch, then it will see all neighbors twice, once over each interface.  The protocol is designed to handle this case gracefully.

I see two solutions to your problem:
1. Use a string of the form `address%interface` as the key, as @PolynomialDivision suggested.
2. Use the address of the `struct neighbor` as the key.  This is the approach used by Babeld's native dumping code:
  https://github.com/jech/babeld/blob/master/local.c#L163

marek22k, 2023-10-04:
Personally, I think the former suggestion is nicer.

stargieg, 2024-07-20:
Hi i am running in the same issue. :-(
my workaround: 
```
ubus call babeld get_routes | \
sed -e 's/^\t\t\(".*"\): {/\t\t\{\n\t\t\t"prefix": \1,/' \
-e 's/^\(\t".*": \){/\1[/' \
-e 's/^\t}/\t]/' \
-e 's/src-prefix/src_prefix/'
```
```
{
	"IPv4": [
		
	],
	"IPv6": [
		{
			"prefix": "2003:ea:d7ff:ad3::/64",
			"src_prefix": "::/0",
			"route_metric": 384,
			"route_smoothed_metric": 384,
			"refmetric": 0,
			"id": "da:58:d7:ff:fe:00:6e:e9",
			"seqno": 51621,
			"age": 2,
			"via": "fe80::da58:d7ff:fe00:6ee8",
			"installed": true,
			"feasible": true
		},
		{
			"prefix": "2003:ea:d70a:fce2::1/128",
			"src_prefix": "::/0",
			"route_metric": 768,
			"route_smoothed_metric": 768,
			"refmetric": 384,
			"id": "66:66:b3:ff:fe:8a:55:fe",
			"seqno": 56351,
			"age": 2,
			"via": "fe80::da58:d7ff:fe00:6ee8",
			"installed": false,
			"feasible": true
		}
	]
}
```
Should i fixit in c? here: https://github.com/openwrt/routing/blob/master/babeld/src/ubus.c
js here: https://github.com/openwrt/luci/blob/master/applications/luci-app-babeld/htdocs/luci-static/resources/babeld.js
doku here: https://openwrt.org/docs/guide-user/services/babeld

stargieg, 2024-08-03:
I'm testing patches:
https://github.com/openwrt/routing/compare/master...stargieg:routing:babeld-ubus
https://github.com/openwrt/luci/compare/master...stargieg:luci:babeld-ubus

Changes:
Use blobmsg_open_array for ip lists
Use underscore in fild name like libubox/jshn.sh
Change format type in luci format()
Remove element Channel in luci. Not in ubus.c
Add element uhello_reach in luci. It is in ubus.c. Don't now it is usefull?
Notes or comments before I make a PR?
@PolynomialDivision 
How can we publish this in the repo luci and routing at the same time?
@jow- 

```
#ubus call babeld get_neighbours
{
	"IPv4": [
		
	],
	"IPv6": [
		{
			"address": "fe80::26a4:3cff:fee5:69dc",
			"dev": "br-lan",
			"hello_reach": 65535,
			"uhello_reach": 0,
			"rxcost": 96,
			"txcost": 96,
			"rtt": "0.000",
			"if_up": true
		},
		{
			"address": "fe80::618:d6ff:fe25:fe5d",
			"dev": "br-lan",
			"hello_reach": 65535,
			"uhello_reach": 0,
			"rxcost": 96,
			"txcost": 96,
			"rtt": "0.000",
			"if_up": true
		},
		{
			"address": "fe80::26a4:3cff:fee5:6fd7",
			"dev": "br-lan",
			"hello_reach": 65535,
			"uhello_reach": 0,
			"rxcost": 96,
			"txcost": 96,
			"rtt": "0.000",
			"if_up": true
		},
		{
			"address": "fe80::26a4:3cff:fee3:7ddb",
			"dev": "br-lan",
			"hello_reach": 65535,
			"uhello_reach": 0,
			"rxcost": 96,
			"txcost": 96,
			"rtt": "0.000",
			"if_up": true
		},
		{
			"address": "fe80::26a4:3cff:febf:12fa",
			"dev": "br-lan",
			"hello_reach": 65535,
			"uhello_reach": 0,
			"rxcost": 96,
			"txcost": 96,
			"rtt": "0.000",
			"if_up": true
		}
	]
}
```

closed 2024-09-08 by PolynomialDivision