openwrt/routing #972: nodogsplash: does not compile

PolynomialDivision, 2023-05-16:
Maintainer: @lynxis @bluewavenet 
Environment: mt7622

Description:

```
rm -f /home/nick/openwrt/build_dir/target-aarch64_cortex-a53_musl/nodogsplash-5.0.0/.built
touch /home/nick/openwrt/build_dir/target-aarch64_cortex-a53_musl/nodogsplash-5.0.0/.built_check
CFLAGS="-Os -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -ffile-prefix-map=/home/nick/openwrt/build_dir/target-aarch64_cortex-a53_musl/nodogsplash-5.0.0=nodogsplash-5.0.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/home/nick/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.3.0_musl/usr/include -I/home/nick/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.3.0_musl/include/fortify -I/home/nick/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.3.0_musl/include " CXXFLAGS="-Os -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -ffile-prefix-map=/home/nick/openwrt/build_dir/target-aarch64_cortex-a53_musl/nodogsplash-5.0.0=nodogsplash-5.0.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/home/nick/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.3.0_musl/usr/include -I/home/nick/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.3.0_musl/include/fortify -I/home/nick/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.3.0_musl/include " LDFLAGS="-L/home/nick/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.3.0_musl/usr/lib -L/home/nick/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.3.0_musl/lib -znow -zrelro " make  -C /home/nick/openwrt/build_dir/target-aarch64_cortex-a53_musl/nodogsplash-5.0.0/. AR="aarch64-openwrt-linux-musl-gcc-ar" AS="aarch64-openwrt-linux-musl-gcc -c -Os -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -ffile-prefix-map=/home/nick/openwrt/build_dir/target-aarch64_cortex-a53_musl/nodogsplash-5.0.0=nodogsplash-5.0.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD=aarch64-openwrt-linux-musl-ld NM="aarch64-openwrt-linux-musl-gcc-nm" CC="aarch64-openwrt-linux-musl-gcc" GCC="aarch64-openwrt-linux-musl-gcc" CXX="aarch64-openwrt-linux-musl-g++" RANLIB="aarch64-openwrt-linux-musl-gcc-ranlib" STRIP=aarch64-openwrt-linux-musl-strip OBJCOPY=aarch64-openwrt-linux-musl-objcopy OBJDUMP=aarch64-openwrt-linux-musl-objdump SIZE=aarch64-openwrt-linux-musl-size CROSS="aarch64-openwrt-linux-musl-" ARCH="aarch64" ;
make[3]: Entering directory '/home/nick/openwrt/build_dir/target-aarch64_cortex-a53_musl/nodogsplash-5.0.0'
aarch64-openwrt-linux-musl-gcc  -Os -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -ffile-prefix-map=/home/nick/openwrt/build_dir/target-aarch64_cortex-a53_musl/nodogsplash-5.0.0=nodogsplash-5.0.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/home/nick/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.3.0_musl/usr/include -I/home/nick/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.3.0_musl/include/fortify -I/home/nick/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.3.0_musl/include  -Isrc -c src/main.c -o src/main.o
src/main.c:65:10: fatal error: microhttpd.h: No such file or directory
   65 | #include <microhttpd.h>
      |          ^~~~~~~~~~~~~~
compilation terminated.
make[3]: *** [Makefile:20: src/main.o] Error 1
```

PolynomialDivision, 2023-05-16:
Sorry, my bad. I just checked out the routing feed and it has dependencies to packages feed. However, it seems like I don't get any warning if a dependency is missing.

closed 2023-05-16 by PolynomialDivision

BKPepe, 2023-05-16:
I am thinking that if the package within routing repository depends on package from another feed, then it should be moved there.

bluewavenet, 2023-05-17:
@PolynomialDivision 
I think ultimately this is because gnutls is currently failing, in turn leading to libmicrohttpd to fail... not 100% sure.