openwrt/routing #994: Quagga: is no longer being developed

marek22k, 2023-07-20:
Maintainer: @BKPepe @PolynomialDivision @neheb

Description:
The latest Quagge version is 1.2.4 from 2018. frr is a kind of successor of Quagga. The website and the mailling list of Quagge is also no longer accessible - only a copy also archive.org. I would suggest removing Quagga from the OpenWrt package sources.

Another question: is it intentional that Frr https://github.com/openwrt/packages/tree/openwrt-22.03/net/frr is in the main repo and not the routing repo?

closed 2023-07-22 by mwarning

PolynomialDivision, 2023-07-23:
If we start removing unmaintained projects than we probably need to remove more. ;)

mwarning, 2023-07-23:
@PolynomialDivision quagga is not only unmaintained, but the sources are not available via conventional channels. So nobody can build it, which also means that there are no users. :/

BKPepe, 2023-07-23:
Guys, yes, both of you are right. In OpenWrt repositories there are plenty of packages, which projects, which are abandoned, not maintained, but they have their purpose. On the other hand, we should prefer quality over quantity, because there are few active contributors to our projects and if they are, then only a few people are reviewing it. 

Removing those packages helps to use less resources and have builds faster, which then could be shipped to users in a short time to keep them security against security vulnerabilities.

But... if we are talking about quagga, it was right thing to do. If you are looking at Debian, then you can find the package only in old stable buster distribution (https://packages.debian.org/buster/quagga) and as well it was removed from Ubuntu 22.04 (source: https://askubuntu.com/questions/1417548/has-quagga-been-removed-from-22-04).