openwrt/routing #1000: error: recursive dependency detected!

hejiawencc, 2023-07-27:
Description:

```
Collecting package info: done
tmp/.config-package.in:92125:error: recursive dependency detected!
tmp/.config-package.in:92125:   symbol PACKAGE_nodogsplash depends on PACKAGE_opennds
tmp/.config-package.in:92144:   symbol PACKAGE_opennds depends on PACKAGE_nodogsplash
For a resolution refer to Documentation/kbuild/kconfig-language.rst
subsection "Kconfig recursive dependency limitations"
```

mwarning, 2023-07-27:
hi @hejiawencc, how can we reproduce this issue? I tried, but wasn't able to.
I build some image with nodogsplash and opennds both selected as module and it worked.
Building an image with nodogsplash only included worked as well.

mwarning, 2023-07-27:
@bluewavenet fyi

bluewavenet, 2023-07-27:
@hejiawencc 
Just giving us an error message from somewhere without any context is not very helpful.
Can you describe what it was you were doing when you saw the error?

mwarning, 2023-07-27:
* https://github.com/openwrt/routing/pull/1002
* https://github.com/openwrt/routing/pull/1001

hejiawencc, 2023-07-28:
Good morning. I'm very sorry that I didn't provide enough useful information, if there is a next time, I will make improvements.

Glad this issue has been resolved by #1002

mwarning, 2023-07-29:
fixed

mwarning, 2023-07-29:
@hejiawencc thank you for reporting this issue. We failed to notice it. :-)

closed 2023-07-29 by mwarning