openwrt/routing #1028: ohybridproxy: dns_sd.h is missing

BKPepe, 2023-10-13:
Maintainer: @sbyx and previously who was active about this package @fingon 
Environment: OpenWrt master, aarch64_cortex-a53

Description:

```
/builder/shared-workdir/build/sdk/build_dir/target-aarch64_cortex-a53_musl/ohybridproxy-2020-05-22-0dfef1eb/src/dns2mdns_i.h:21:10: fatal error: dns_sd.h: No such file or directory
   21 | #include <dns_sd.h>
      |          ^~~~~~~~~~
compilation terminated.
ninja: build stopped: subcommand failed.
make[3]: *** [Makefile:55: /builder/shared-workdir/build/sdk/build_dir/target-aarch64_cortex-a53_musl/ohybridproxy-2020-05-22-0dfef1eb/.built] Error 1
time: package/feeds/routing/ohybridproxy/compile#0.91#0.60#1.95
````

Snip from https://downloads.openwrt.org/snapshots/faillogs/aarch64_cortex-a53/routing/ohybridproxy/compile.txt

BKPepe, 2023-10-19:
same as https://github.com/openwrt/routing/issues/1029

fingon, 2023-10-19:
cut-n-pasting here too - fwiw the problem is this: https://github.com/openwrt/packages/commit/e0c85eb48556798af5ef6a2b2f7e1afdf9880163 and adding it to include path would probably fix it. As the location is nonstandard, I'm not changing the package itself at least.

BKPepe, 2023-10-27:
ping @neheb

neheb, 2023-10-29:
That commit should be reverted as it’s clearly wrong.

the issue is that both avahi-compat-dnssd and mdnaresponder install to the same location and therefore conflict.

The ideal solution is to get rid of mdnsresponder and only use avahi.

@stintel seems to be set on using mdnsresponder. Which probably means getting rid of avahi’s compat. I tried and failed to have them coexist.

stintel, 2023-10-29:
> The ideal solution is to get rid of mdnsresponder and only use avahi.

Avahi and its DBUS dependency is not something I feel comfortable with recommending people run on a router.

stintel, 2023-10-29:
openwrt/packages@a6d91c989d30a0668809f16742c0e19371541872

closed 2023-10-29 by stintel