openwrt/routing #1044: batman-adv: `proto_batadv_vlan_teardown` not implemented in `batadv_vlan.sh`

raenye, 2024-02-04:
Maintainer: @simonwunderlich 
Environment: multiple (mediatek/fiilogic, mediatek/mt7622, ramips/mt7621, ath79/generic

Description:
There is no implementation of `proto_batadv_hardif_teardown` in `files/lib/netifd/proto/batadv_vlan.sh`, which causes the following error in the log:
```
daemon.notice netifd: batmesh1 (18940): ./batadv_vlan.sh: eval: line 37: proto_batadv_vlan_teardown: not found
```

Not sure if something really needs to be done, or it's just log litter.
Thanks!

Relevant part of `/etc/config/network`:
```
config interface 'bat0'
        option proto 'batadv'
        option routing_algo 'BATMAN_IV'
        option aggregated_ogms '1'
        option bonding '1'
        option bridge_loop_avoidance '1'
        option gw_mode 'server'
        option gw_bandwidth '1000mbit/100mbit'
        option hop_penalty '30'
        option network_coding '1'

config interface 'batmesh0'
        option proto 'batadv_hardif'
        option master 'bat0'
        option mtu '2304'
        option delegate '0'

config interface 'batmesh1'
        option proto 'batadv_hardif'
        option master 'bat0'
        option mtu '2304'
        option delegate '0'

config interface 'bat0-lan'
    option proto 'batadv_vlan'
    option ifname 'bat0.10'

config interface 'bat0-guest'
    option proto 'batadv_vlan'
    option ifname 'bat0.20'

config interface 'bat0-iot'
    option proto 'batadv_vlan'
    option ifname 'bat0.30'

config device
        option name 'br-lan'
        option type 'bridge'
        list ports 'lan1'
        list ports 'lan2'
        list ports 'lan3'
        list ports 'bat0-lan'

config device
        option name 'br-guest'
        option type 'bridge'
        list ports 'bat0-guest'

config device
        option name 'br-iot'
        option type 'bridge'
        list ports 'bat0-iot'
```

ecsv, 2024-02-05:
Thanks for the ticket. Also saw this after updating some of our office APs to 23.05.

There is nothing to be done by the proto. So I will simply add a stub function.

closed 2024-02-05 by simonwunderlich

raenye, 2024-02-05:
Thanks for the quick fix.
BTW, now that I read the code, does it even make sense to use `batadv_vlan` for VLANs without AP isolation? does it serve to solve problems such as https://forum.openwrt.org/t/batman-log-spam-from/168516?

ecsv, 2024-02-05:
> Thanks for the quick fix. BTW, now that I read the code, does it even make sense to use `batadv_vlan` for VLANs without AP isolation?
> does it serve to solve problems such as https://forum.openwrt.org/t/batman-log-spam-from/168516?

Yes and no.

It doesn't have any direct affect by toggling the ap_isolate setting. But netifd should automatically create the VLAN in this case - just because you reference a VLAN as `ifname`.  And if something created the VLAN then batman-adv will no longer complain about "non-existent VLAN"

raenye, 2024-02-05:
But adding `bat.X` to a bridge would serve the same purpose wrt `netifd`, won't it?

PR #1045 (2024-02-05): batman-adv: Provide teardown hook for batadv_vlan proto
Maintainer: @simonwunderlich 
Compile tested: ath79, GL.iNet GL-AR750
Run tested: ath79, GL.iNet GL-AR750

Description:

The batadv_vlan proto doesn't need to do anything when it gets teared down. But the scripts are still trying to call the teardown function of this proto. This results in warnings like:

    daemon.notice netifd: batmesh1 (18940): ./batadv_vlan.sh: eval: line 37: proto_batadv_vlan_teardown: not found

Just providing a stub function avoids this log spam.

Fixes: #1044
Reported-by: Rani Hod <rani.hod@gmail.com>
Fixes: f520 [... cut]
commit 9090aefa (2024-02-05): batman-adv: Provide teardown hook for batadv_vlan proto

The batadv_vlan proto doesn't need to do anything when it gets teared down.
But the scripts are still trying to call the teardown function of this
proto. This results in warnings like:

    daemon.notice netifd: batmesh1 (18940): ./batadv_vlan.sh: eval: line 37: proto_batadv_vlan_teardown: not found

Just providing a stub function avoids this log spam.

Fixes: #1044
Reported-by: Rani Hod <rani.hod@gmail.com>
Fixes: f5205d7d2434 ("batman-adv: upgrade package to latest release 2014.2.0")
PR #1046 (2024-02-05): openwrt-23.05: batman-adv: Provide teardown hook for batadv_vlan proto
Maintainer: @simonwunderlich 
Compile tested: ath79, GL.iNet GL-AR750
Run tested: ath79, GL.iNet GL-AR750

Description:

The batadv_vlan proto doesn't need to do anything when it gets teared down. But the scripts are still trying to call the teardown function of this proto. This results in warnings like:

    daemon.notice netifd: batmesh1 (18940): ./batadv_vlan.sh: eval: line 37: proto_batadv_vlan_teardown: not found

Just providing a stub function avoids this log spam.

Fixes: #1044
Reported-by: Rani Hod <rani.hod@gmail.c [... cut]
commit 5a7cac9b (2024-02-05): batman-adv: Provide teardown hook for batadv_vlan proto

The batadv_vlan proto doesn't need to do anything when it gets teared down.
But the scripts are still trying to call the teardown function of this
proto. This results in warnings like:

    daemon.notice netifd: batmesh1 (18940): ./batadv_vlan.sh: eval: line 37: proto_batadv_vlan_teardown: not found

Just providing a stub function avoids this log spam.

Fixes: #1044
Reported-by: Rani Hod <rani.hod@gmail.com>
Fixes: f5205d7d2434 ("batman-adv: upgrade package to latest release 2014.2.0")