openwrt/routing #1102: bird3 3.0.1 produces invalid opcodes on amd64

jmarcet, 2025-01-18:
I need to revert 9935dfd to have a working bird3 on master amd64.

Version 3.0.1 fails due to invalid opcodes:
```sh
❯ dmesg | grep bird
[   42.148485] traps: bird[15245] trap invalid opcode ip:5607fcb20065 sp:7ffc83b7d1e0 error:0 in bird[5607fcac3000+108000]
[   49.147687] traps: bird[9477] trap invalid opcode ip:55df30d52065 sp:7ffcae7ad480 error:0 in bird[55df30cf5000+108000]
[   73.157254] traps: bird[9782] trap invalid opcode ip:55ebcc7d3065 sp:7ffc3c37ed60 error:0 in bird[55ebcc776000+108000]
[  104.148641] traps: bird[14850] trap invalid opcode ip:56193e555065 sp:7ffcaa897f20 error:0 in bird[56193e4f8000+108000]
[  133.158205] traps: bird[28945] trap invalid opcode ip:56391d8f7065 sp:7fffa148f360 error:0 in bird[56391d89a000+108000]
[  140.148007] traps: bird[31186] trap invalid opcode ip:55b27e672065 sp:7ffc5c24d780 error:0 in bird[55b27e615000+108000]
```

PolynomialDivision, 2025-01-18:
@jmarcet Please report that to the bird mailinglist:
https://bird.network.cz/mailman/listinfo/bird-users

BKPepe, 2025-05-14:
I think the issue here is that recent PRs for bird are not tested on the real device and it counts only on CI here.

BKPepe, 2025-05-17:
I've looked at it further. Version 3.0.1 is no longer in the master branch as there is version 3.1.0 already.
However, we do have 3.0.1 in the stable version (OpenWrt 24.10), even though there is already version 3.0.3. @jmarcet would you be so kind to test the 3.0.3 to verify if your issue is fixed, please?

jmarcet, 2025-05-17:
At the moment I'm reverting 3.10 and 3.0.1 since both fail. I have not tried 3.0.3. Shall I try it?

BKPepe, 2025-05-17:
Please do.

jmarcet, 2025-05-17:
I cannot find version 3.0.3:
[https://bird.network.cz/download/](https://bird.network.cz/download/)

[image]

BKPepe, 2025-05-17:
Hmm, but it is released based on this email:
https://trubka.network.cz/pipermail/bird-users/2025-May/018199.html

jmarcet, 2025-05-17:
> Hmm, but it is released based on this email: https://trubka.network.cz/pipermail/bird-users/2025-May/018199.html

```diff
diff --git a/bird3/Makefile b/bird3/Makefile
index f7a36f0..7efadac 100644
--- a/bird3/Makefile
+++ b/bird3/Makefile
@@ -3,12 +3,12 @@
 include $(TOPDIR)/rules.mk

 PKG_NAME:=bird3
-PKG_VERSION:=3.0.0
+PKG_VERSION:=3.0.3
 PKG_RELEASE:=1

 PKG_SOURCE:=bird-$(PKG_VERSION).tar.gz
-PKG_SOURCE_URL:=https://bird.network.cz/download/
-PKG_HASH:=8130440a2e273ba6456df2fb3acb43da7cb4d566f94a294a3a52a1b118f2512a
+PKG_SOURCE_URL:=https://bird.nic.cz/download/
+PKG_HASH:=8328499b25a21e466c8aed5d71f00a2a71551c1d4afb21b5fd57d0081943479b

 PKG_MAINTAINER:=Toke Høiland-Jørgensen <toke@toke.dk>, Nick Hainke <vincent@systemli.org>
 PKG_LICENSE:=GPL-2.0-or-later
```

jmarcet, 2025-05-17:
I'm afraid 3.0.3 also fails:

```sh
❯ dmesg | grep bird
[505552.562016] traps: bird[14017] trap invalid opcode ip:560a886a75c5 sp:7ffe3326b3c0 error:0 in bird[560a8864a000+109000]
```