openwrt/routing #1118: batmand: fails to compile

BKPepe, 2025-05-23:
Maintainer: @ecsv 
OpenWrt snapshot, aarch64, cortex-a53

Log:
```
hna.c: In function 'hna_local_task_add_ip':
hna.c:97:103: error: implicit declaration of function 'strerror'; did you mean 'perror'? [-Wimplicit-function-declaration]
   97 |                 debug_output(0, "Error - could not lock hna_chg_list mutex in %s(): %s \n", __func__, strerror(errno));
      |                                                                                                       ^~~~~~~~
      |                                                                                                       perror
hna.c: In function 'hna_local_task_add_str':
hna.c:112:26: error: implicit declaration of function 'strchr' [-Wimplicit-function-declaration]
  112 |         if ((slash_ptr = strchr(hna_string, '/')) == NULL) {
      |                          ^~~~~~
hna.c:31:1: note: include '<string.h>' or provide a declaration of 'strchr'
   30 | #include <arpa/inet.h>
  +++ |+#include <string.h>
   31 | 
hna.c:112:26: warning: incompatible implicit declaration of built-in function 'strchr' [-Wbuiltin-declaration-mismatch]
  112 |         if ((slash_ptr = strchr(hna_string, '/')) == NULL) {
      |                          ^~~~~~
hna.c:112:26: note: include '<string.h>' or provide a declaration of 'strchr'
hna.c: In function 'hna_local_buffer_fill':
hna.c:187:17: error: implicit declaration of function 'memmove' [-Wimplicit-function-declaration]
  187 |                 memmove(&hna_buff_local[num_hna_local * 5], (unsigned char *)&hna_local_entry->addr, 4);
      |                 ^~~~~~~
hna.c:187:17: note: include '<string.h>' or provide a declaration of 'memmove'
hna.c:187:17: warning: incompatible implicit declaration of built-in function 'memmove' [-Wbuiltin-declaration-mismatch]
hna.c:187:17: note: include '<string.h>' or provide a declaration of 'memmove'
hna.c: In function 'hna_buff_delete':
hna.c:480:25: warning: incompatible implicit declaration of built-in function 'memmove' [-Wbuiltin-declaration-mismatch]
  480 |                         memmove(&buf[i], &buf[num_elements - 1], sizeof(struct hna_element));
      |                         ^~~~~~~
hna.c:480:25: note: include '<string.h>' or provide a declaration of 'memmove'
make[4]: *** [Makefile:90: hna.o] Error 1
make[4]: Leaving directory '/builder/shared-workdir/build/sdk/build_dir/target-aarch64_cortex-a53_musl/batmand-1440'
```

Log is located here:
https://downloads.openwrt.org/snapshots/faillogs/aarch64_cortex-a53/routing/batmand/compile.txt

ecsv, 2025-05-23:
@BKPepe I am neither downstream nor upstream maintainer.

Maintainer: @elektra42

ecsv, 2025-05-23:
Looks to me like you just need to add

```
#include <string.h>
```

to hna.c

elektra42, 2025-05-23:
> @BKPepe @elektra42 , you can decide which of the two changes you want:
>
> * upgrade:
>   - [0001-batmand-Upgrade-to-latest-git-snapshot.txt](https://github.com/user-attachments/files/20407858/0001-batmand-Upgrade-to-latest-git-snapshot.txt)
> * removal
>   - [0001-batmand-Drop-package.txt](https://github.com/user-attachments/files/20407871/0001-batmand-Drop-package.txt)
>   - [0002-vis-Drop-package.txt](https://github.com/user-attachments/files/20407874/0002-vis-Drop-package.txt)
> * fix
>   - [0001-batmand-Fix-FTBFS-with-recent-musl.txt](https://github.com/user-attachments/files/20407888/0001-batmand-Fix-FTBFS-with-recent-musl.txt)

Hi Sven and all –

thanks for the help and notification. 

I'd pick this change: 

* fix
 - [0001-batmand-Fix-FTBFS-with-recent-musl.txt](https://github.com/user-attachments/files/20407888/0001-batmand-Fix-FTBFS-with-recent-musl.txt)

I appreciate your feedback.

Best regards,
Elektra

ecsv, 2025-06-20:
Should be fixed via https://github.com/openwrt/routing/commit/d08b1447ba08ea894b76175ec784fb32aae359ec - even when then a full upgrade was performed in https://github.com/openwrt/routing/pull/1130 (which has basically the same effect).